Xilinx/mlir-aie #1556: MLIR::loopUnrollByFactor fails for small loops

AndraBisca, 2024-06-14:
After #1545 designs with loops that have less iterations than the unroll factor fail silently.

The example [here](https://github.com/Xilinx/mlir-aie/blob/unroll-bug/test/objectFifo-stateful-transform/loop_test_small.mlir) shows this error and expected output.

@jtuyls

makslevental, 2024-06-14:
You can use this to compute the trip count of an `scf.for` with static bounds and step: https://github.com/llvm/llvm-project/blob/main/mlir/include/mlir/Dialect/Utils/StaticValueUtils.h#L175.

makslevental, 2024-06-14:
```cpp
          if (!forLoop.getSingleLowerBound() ||
              !forLoop.getSingleUpperBound() || !forLoop.getSingleStep() ||
              constantTripCount(*forLoop.getSingleLowerBound(),
                                *forLoop.getSingleUpperBound(),
                                *forLoop.getSingleStep())
                      .value_or(0) >= unrollFactor)
            forLoop.emitOpError()
                << "could not be unrolled because tripcount < unrollFactor: "
                << unrollFactor << "\n"
```

PR #1568 (2024-08-07): Fully unroll for-loops with tripCount < unrollFactor
Fixes #1556
closed 2024-08-07 by AndraBisca