Xilinx/mlir-aie #1564: XCLBinGen silently emits at most 5 `bo`s in `kernels.json` even if `@sequence` has more

makslevental, 2024-06-14:
https://github.com/Xilinx/mlir-aie/blob/main/test/aie2xclbin/buffers_xclbin.mlir#L115:

```
func.func @sequence(%arg0: memref<1024xi32>, %arg1: memref<1024xi32>, %arg2: memref<1024xi32>, %arg3: memref<1024xi32>, %arg4: memref<1024xi32>, %arg5: memref<1024xi32>)
```

but final bo in kernels.json

```
// CHECK:           {
// CHECK:             "address-qualifier": "GLOBAL",
// CHECK:             "memory-connection": "HOST",
// CHECK:             "name": "bo4",
// CHECK:             "offset": "0x34",
// CHECK:             "type": "void*"
// CHECK:           }
```

cc @nirvedhmeshram

fifield, 2024-06-14:
I've reverted the related changes from https://github.com/Xilinx/mlir-aie/pull/1517 in https://github.com/Xilinx/mlir-aie/pull/1565 so that at least the test makes sense, but it does not solve the complaint in the issue title about no error.

makslevental, 2024-06-15:
> does not solve the complaint in the issue title about no error.

So then why not solve it by checking how many args `sequence` expects...? Or in fact actually emitting the right number of bo entries in the json.

fifield, 2024-06-15:
> > does not solve the complaint in the issue title about no error.
> 
> So then why not solve it by checking how many args `sequence` expects...? Or in fact actually emitting the right number of bo entries in the json instead of hard coding some arbitrary quantity.

Yes that would be the right thing to for someone to implement.

closed 2025-03-25 by jgmelber