Xilinx/mlir-aie #1567: `AIE2TargetModel::isLegalMemtileConnection` incorrect for many `Ctrl`, `DMA` and `Trace` connections

makslevental, 2024-06-16:
```
isLegalMemtileConnection wrong (reports true when false): src: Ctrl0, dst: DMA0
isLegalMemtileConnection wrong (reports true when false): src: Ctrl0, dst: DMA1
isLegalMemtileConnection wrong (reports true when false): src: Ctrl0, dst: DMA2
isLegalMemtileConnection wrong (reports true when false): src: Ctrl0, dst: DMA3
isLegalMemtileConnection wrong (reports true when false): src: Ctrl0, dst: DMA4
isLegalMemtileConnection wrong (reports true when false): src: DMA0, dst: DMA1
isLegalMemtileConnection wrong (reports true when false): src: DMA0, dst: DMA2
isLegalMemtileConnection wrong (reports true when false): src: DMA0, dst: DMA3
isLegalMemtileConnection wrong (reports true when false): src: DMA0, dst: DMA4
isLegalMemtileConnection wrong (reports true when false): src: DMA0, dst: DMA5
isLegalMemtileConnection wrong (reports true when false): src: DMA1, dst: DMA0
isLegalMemtileConnection wrong (reports true when false): src: DMA1, dst: DMA2
isLegalMemtileConnection wrong (reports true when false): src: DMA1, dst: DMA3
isLegalMemtileConnection wrong (reports true when false): src: DMA1, dst: DMA4
isLegalMemtileConnection wrong (reports true when false): src: DMA1, dst: DMA5
isLegalMemtileConnection wrong (reports true when false): src: DMA2, dst: DMA0
isLegalMemtileConnection wrong (reports true when false): src: DMA2, dst: DMA1
isLegalMemtileConnection wrong (reports true when false): src: DMA2, dst: DMA3
isLegalMemtileConnection wrong (reports true when false): src: DMA2, dst: DMA4
isLegalMemtileConnection wrong (reports true when false): src: DMA2, dst: DMA5
isLegalMemtileConnection wrong (reports true when false): src: DMA3, dst: DMA0
isLegalMemtileConnection wrong (reports true when false): src: DMA3, dst: DMA1
isLegalMemtileConnection wrong (reports true when false): src: DMA3, dst: DMA2
isLegalMemtileConnection wrong (reports true when false): src: DMA3, dst: DMA4
isLegalMemtileConnection wrong (reports true when false): src: DMA3, dst: DMA5
isLegalMemtileConnection wrong (reports true when false): src: DMA4, dst: DMA0
isLegalMemtileConnection wrong (reports true when false): src: DMA4, dst: DMA1
isLegalMemtileConnection wrong (reports true when false): src: DMA4, dst: DMA2
isLegalMemtileConnection wrong (reports true when false): src: DMA4, dst: DMA3
isLegalMemtileConnection wrong (reports true when false): src: DMA4, dst: DMA5
isLegalMemtileConnection wrong (reports true when false): src: DMA5, dst: DMA0
isLegalMemtileConnection wrong (reports true when false): src: DMA5, dst: DMA1
isLegalMemtileConnection wrong (reports true when false): src: DMA5, dst: DMA2
isLegalMemtileConnection wrong (reports true when false): src: DMA5, dst: DMA3
isLegalMemtileConnection wrong (reports true when false): src: DMA5, dst: DMA4
isLegalMemtileConnection wrong (reports true when false): src: Trace0, dst: Ctrl0
isLegalMemtileConnection wrong (reports true when false): src: Trace0, dst: DMA0
isLegalMemtileConnection wrong (reports true when false): src: Trace0, dst: DMA1
isLegalMemtileConnection wrong (reports true when false): src: Trace0, dst: DMA2
isLegalMemtileConnection wrong (reports true when false): src: Trace0, dst: DMA3
isLegalMemtileConnection wrong (reports true when false): src: Trace0, dst: DMA4
isLegalMemtileConnection wrong (reports true when false): src: Trace0, dst: North0
isLegalMemtileConnection wrong (reports true when false): src: Trace0, dst: North1
isLegalMemtileConnection wrong (reports true when false): src: Trace0, dst: North2
isLegalMemtileConnection wrong (reports true when false): src: Trace0, dst: North3
isLegalMemtileConnection wrong (reports true when false): src: Trace0, dst: North4
isLegalMemtileConnection wrong (reports true when false): src: Trace0, dst: North5
```

See Figure 6-9 in arch spec.

AndraBisca, 2024-07-10:
Is this issue ready to close @Yu-Zhewen ?

erwei-xilinx, 2024-07-16:
Yeah this issue can be closed, as the new AIETargetModel is verified against all cases above, with https://github.com/Xilinx/mlir-aie/pull/1602.

closed 2024-07-16 by erwei-xilinx