Xilinx/mlir-aie #1593: Repeat buffer failed

ngdymx, 2024-07-02:
Hi team,

My purpose is to repeat a memref<64xi8> eight times to form a memref<512xi8> on chip with objectfifo. I write a mlir code shown in the following. 

```
module {
  aie.device(npu1_1col) {
    %tile_0_0 = aie.tile(0, 0)
    %tile_0_2 = aie.tile(0, 2)
    aie.objectfifo @in0(%tile_0_0, {%tile_0_2}, 2 : i32) : !aie.objectfifo<memref<8xi8>>
    aie.objectfifo @out0(%tile_0_2, {%tile_0_0}, 2 : i32) : !aie.objectfifo<memref<64xi8>>

    %core_0_2 = aie.core(%tile_0_2) {
      %c0 = arith.constant 0 : index
      %c9223372036854775807 = arith.constant 9223372036854775807 : index
      %c1 = arith.constant 1 : index
      %c2 = arith.constant 2 : index
      %c8 = arith.constant 8 : index

      scf.for %arg0 = %c0 to %c9223372036854775807 step %c1 {

        %0 = aie.objectfifo.acquire @out0(Produce, 1) : !aie.objectfifosubview<memref<64xi8>>
        %1 = aie.objectfifo.subview.access %0[0] : !aie.objectfifosubview<memref<64xi8>> -> memref<64xi8>

        scf.for %arg1 = %c0 to %c8 step %c1 {
          %2 = aie.objectfifo.acquire @in0(Consume, 1) : !aie.objectfifosubview<memref<8xi8>>
          %3 = aie.objectfifo.subview.access %2[0] : !aie.objectfifosubview<memref<8xi8>> -> memref<8xi8>

          %c9 = arith.muli %arg1, %c8 : index

          scf.for %arg2 = %c0 to %c8 step %c1 {
            %4 = memref.load %3[%arg2] : memref<8xi8>
            %5 = arith.addi %arg2, %c9 : index
            memref.store %4, %1[%5] : memref<64xi8>
          }
          aie.objectfifo.release @in0(Consume, 1)
        }
        aie.objectfifo.release @out0(Produce, 1)
      }
      aie.end
    }
    func.func @sequence(%arg0: memref<2xi32>, %arg1: memref<2xi32>, %arg2: memref<16xi32>) {
      aiex.npu.dma_memcpy_nd(0, 0, %arg2[0, 0, 0, 0][1, 1, 1, 16][0, 0, 0]) {id = 0 : i64, metadata = @out0} : memref<16xi32>
      aiex.npu.dma_memcpy_nd(0, 0, %arg0[0, 0, 0, 0][8, 1, 1, 2][0, 0, 0]) {id = 1 : i64, metadata = @in0} : memref<2xi32>
      aiex.npu.sync {channel = 0 : i32, column = 0 : i32, column_num = 1 : i32, direction = 0 : i32, row = 0 : i32, row_num = 1 : i32}
      return
    }
  }
}
```

In this code, the ```%c8 ```  is the replicate time, when I compile it, it gives me a "WorkerSegmentation fault" error. 

I tried to reduce the replicate time to 2 by replacing ```%c8``` with ```%c2```,  and it works.  The result is shown below: 

```
Input:
[9 1 3 7 8 7 8 2]
Output:
[[9 1 3 7 8 7 8 2]
 [9 1 3 7 8 7 8 2]
 [0 0 0 0 0 0 0 0]
 [0 0 0 0 0 0 0 0]
 [0 0 0 0 0 0 0 0]
 [0 0 0 0 0 0 0 0]
 [0 0 0 0 0 0 0 0]
 [0 0 0 0 0 0 0 0]]
```

May I ask if it is a bug of the compiler?

AndraBisca, 2024-07-03:
Hello and thank you for signaling this issue!

Would you be able to provide us with your test file, as well as with the build and run commands (or the Makefile) that you used?

ngdymx, 2024-07-03:
Hi,

Sure.

The file structure is shown below.  The  ``` test.py```  is in the host folder,  and  ```aie.mlir```  is in the kernel folder.
[image]

```
test.py:

import torch
import torch.nn as nn
import sys
import math
from aie.utils.ml import DataShaper
import time
import os
import numpy as np
from aie.utils.xrt import setup_aie, extract_trace, write_out_trace, execute
import aie.utils.test as test_utils
import einops

torch.use_deterministic_algorithms(True)
torch.manual_seed(0)

def main(opts):
    design = "repeat_buffer"
    xclbin_path = opts.xclbin
    insts_path = opts.instr

    log_folder = "log/"
    if not os.path.exists(log_folder):
        os.makedirs(log_folder)

    num_iter = 1
    npu_time_total = 0
    npu_time_min = 9999999
    npu_time_max = 0
    
    dtype_in = np.dtype("int8")
    dtype_out = np.dtype("int8")

    shape_in_act = (8, )
    shape_out = (8, 8)

    # ------------------------------------------------------
    # Initialize activation, weights, scaling factor for int8 model
    # ------------------------------------------------------
    int_inp = torch.randint(1, 10, (8, )).type(torch.FloatTensor)
    
    # ------------------------------------------------------
    # Get device, load the xclbin & kernel and register them
    # ------------------------------------------------------
    app = setup_aie(
        xclbin_path,
        insts_path,
        shape_in_act,
        dtype_in,
        shape_in_act,
        dtype_in,
        shape_out,
        dtype_out,
    )
    # ------------------------------------------------------
    # Reorder input data-layout
    # ------------------------------------------------------
    h = 8
    golden_input = einops.repeat(int_inp, "h -> d h", d = 8).clone().contiguous()
    
    before_input = int_inp.squeeze().data.numpy().astype(dtype_in)
[... 207 lines cut ...]
    ${Boost_INCLUDE_DIRS}
)

target_link_directories(${currentTarget} PUBLIC
    ${XRT_LIB_DIR}
    ${OpenCV_LIB_PATH}
    ${Boost_LIBRARY_DIRS}
)

if (NOT WSL)
    target_link_libraries(${currentTarget} PUBLIC
        xrt_coreutil
        ${OpenCV_LIBS}
        boost_program_options
        boost_filesystem
    )
else()
    target_link_libraries(${currentTarget} PUBLIC
        xrt_coreutil
        ${OpenCV_LIBS}
    )
endif()
```

``make bitstream``` is to generate ```xclbin``` file and ```make run_py``` is testing. Thank you very much!

AndraBisca, 2024-07-11:
Hello again! Thank you for your patience.

I tried running your design locally on the most recent version of the `main` branch with the files that you have provided and the test passes for me, with the expected output. The only thing I changed was in the MLIR source file: lines 38 and 39, I changed the strides to `[0, 0, 0, 1]`. This is also the expected format when writing the design using python bindings that will then be lowered to MLIR. Does this change work for you as well?

Please let me know if this issue persists!

ngdymx, 2024-07-11:
Hi,

Thank you very much for your help.

I try it and it gives me an ```'aiex.npu.dma_memcpy_nd' op attribute 'static_strides' failed to satisfy constraint: i64 dense array attribute with exactly 3 elements``` error. 

So, I think I need to update the version.

Thank you very much!

AndraBisca, 2024-07-11:
It does indeed look like a version issue, I had the inverse problem:
`'aiex.npu.dma_memcpy_nd' op attribute 'static_strides' failed to satisfy constraint: i64 dense array attribute with exactly 4 elements`

ngdymx, 2024-07-11:
lol, thank you very much.

I will update the version and try it again.

ngdymx, 2024-07-13:
Hello again,

It works, thank you very much!

By the way, I found that in the old version, it took around 1 second to run the ```resnet``` under ```programming_example/ml/resnet```.  Now, it only takes around 4 ms to run the same project.

May I ask how is it achieved and how to interoperate the time consumption?

AndraBisca, 2024-07-23:
Hello again,

I believe this was the result of a number of fixes and improvements on the XRT side, and not because of anything that we changed in this repo.

If everything is now working on your end is this issue alright to close?

ngdymx, 2024-07-23:
Hi, 

Yes, thank you very much!

closed 2024-07-23 by ngdymx