Xilinx/mlir-aie #1743: ISSUE: ERROR when using shared buffer and lock function

ngdymx, 2024-09-04:
Hi team,

I am working on using ```aie.use_lock``` to control the write and read sequence for sharing a buffer between two ComputeTiles. However, I’ve encountered an issue that I’m unsure how to resolve.

Could you please assist me with this? I would greatly appreciate your help.

I have configured the ComputeTiles (1, 4) and (0, 4) as follows: the ```Add one function``` is implemented on tile (1, 4), while the ```passthrough function``` is placed on tile (0, 4). The attached figure illustrates the dataflow diagram for this setup.
[image]

I mimic the ```aie.mlir``` under the folder ```/mlir-aie/mlir_tutorials/tutorial-3``` to use the ```aie.use_lock```.

```mlir
%core_1_4 = aie.core(%tile_1_4) {      
      // while (1)                                                                                                                                      
      scf.for %arg0 = %c0 to %c9223372036854775807 step %c1  {         
      // for i in range(4):                                                                                                                                                                 
             scf.for %arg1 = %c0_0 to %c4 step %c1_1{                                                                                                                                 
                  %0 = aie.objectfifo.acquire @in(Consume, 1) : !aie.objectfifosubview<memref<128xi32>>                                                                                  
                  %1 = aie.objectfifo.subview.access %0[0] : !aie.objectfifosubview<memref<128xi32>> -> memref<128xi32>     
                  //                                                             
                  aie.use_lock(%lock04, "Acquire", 0)                                                                                                                                                                                                                                                           
                  func.call @addOne(%1, %shareBuffer, %c128_i32) : (memref<128xi32>, memref<128xi32>, i32) -> ()                                                                         
                  aie.use_lock(%lock04, "Release", 1)     
                  //                                                                                                                               
                  aie.objectfifo.release @in(Consume, 1)                                                                                                                                 
           }                                                                                                                                                                         
    } 
                                                                                                                                         
%core_0_4 = aie.core(%tile_0_4) {   
     // while (1)                                                                                                                                     
     scf.for %arg0 = %c0 to %c9223372036854775807 step %c1 {    
     // for i in range(4):                                                                
           scf.for %arg1 = %c0_0 to %c4 step %c1_1{                                                                                                                                 
                 %0 = aie.objectfifo.acquire @out(Produce, 1) : !aie.objectfifosubview<memref<128xi32>>                                                                                 
                 %1 = aie.objectfifo.subview.access %0[0] : !aie.objectfifosubview<memref<128xi32>> -> memref<128xi32>   
                 //
                 aie.use_lock(%lock04, "Acquire", 1)                                                                                                                                    
                 memref.copy %shareBuffer, %1 : memref<128xi32> to memref<128xi32>                                                                                                                  
                 aie.use_lock(%lock04, "Release", 0)  
                 //                                                                                                                                  
                 aie.objectfifo.release @out(Produce, 1)                                                                                                                                                                                                                                                                                                                
           }                                                                                                                                                                        
   }
```

The input data is: 
[image]

The output I am expecting to see is as follows:
[image]

The output I got:
[image]

After obtaining this result, I modified the``` passthrough function``` on tile (0, 4) to an ```add_i function```, where ```i``` is the index parameter of the for loop, the range of ```i``` is 1 to 4 step 1. This change was made to test the running time of tile (0, 4).
[image]

The result I got:
[image]

I have confirmed that the running time for tile (0, 4) is 4, which matches our expectations.

Then,  I modified the```add one function``` on tile (1, 4) to an ```add_i function``` to test the running time of tile (1, 4).
[image]

The result I got:
[image]

In this case, the running time for tile (1, 4) appears to occur only once. However, since tile (0, 4) operates successfully, the ```aie.use_lock``` acquire and release operations on tile (1, 4) should also function correctly. This suggests that the lock acquire/release on tile (1, 4) are actually ran for four times, but the function (kernel) is only being called once. 

[... 354 lines cut ...]
	mkdir -p ${@D}
	cd ${@D} && xchesscc_wrapper ${CHESSCCWRAP2_FLAGS} -DINT8_ACT -c $< -o ${@F}

# Build bitstream
${XCLBIN_TARGET}: ${MLIR_TARGET} 
	mkdir -p ${@D}
	cd ${BITSTREAM_O_DIR} && aiecc.py --aie-generate-cdo --no-compile-host --xclbin-name=${@F} \
		--aie-generate-npu --npu-insts-name=${INSTS_TARGET:${BITSTREAM_O_DIR}/%=%} $(<:${MLIR_O_DIR}/%=../../kernel/%) 
	#cd ${@D} && aiecc.py --aie-generate-cdo --no-compile-host --xclbin-name=${@F} \
	#			--aie-generate-npu --npu-insts-name=insts.txt $(<:%=../%)

.PHONY: run
run: ${HOST_C_TARGET} ${XCLBIN_TARGET} ${INSTS_TARGET} #sign
	export XRT_HACK_UNSECURE_LOADING_XCLBIN=1 && \
	./$< -x ${XCLBIN_TARGET} -i ${INSTS_TARGET} -k MLIR_AIE

trace: ${HOST_C_TARGET} ${XCLBIN_TARGET} ${INSTS_TARGET} # sign
	export XRT_HACK_UNSECURE_LOADING_XCLBIN=1 && \
	./$< -x ${XCLBIN_TARGET} -i ${INSTS_TARGET} -k MLIR_AIE -t ${trace_size}
	./parse_trace.py --filename trace.txt --mlir ${MLIR_TARGET} --colshift 1 > trace_mm.json

run_py: ${XCLBIN_TARGET} ${INSTS_TARGET} ${SRCDIR}/host/test.py
	python3 ${SRCDIR}/host/test.py -x ${<} -i ${INSTS_TARGET} -k MLIR_AIE

```

AndraBisca, 2024-11-17:
Hey @ngdymx, thank you for filling this issue! Your testing so far is great to help us isolate the problem.

Could you try instead of using a single lock for your shared buffer to use two locks: a producer lock which is initialized with 1, and a consumer lock initialized with 0. I am suspicious that the lock pattern you tried only works for the AIE1 architecture and not AIE2. The use_lock calls will need to be updated as well.

Here is how that would look:
```
% lock_a04_prod = aie.lock(%tile_0_4, 0) {init = 1 : i32, sym_name = "lock_a04_prod"}
% lock_a04_cons = aie.lock(%tile_0_4, 1) {init = 0 : i32, sym_name = "lock_a04_cons"}
// on %tile_1_4:
aie.use_lock(%lock_a04_prod, AcquireGreaterEqual, 1)
// function call
aie.use_lock(%lock_a04_cons, Release, 1)
        
// on %tile_1_4:
aie.use_lock(%lock_a04_cons, AcquireGreaterEqual, 1)
// function call
aie.use_lock(%lock_a04_prod, Release, 1)
```

AndraBisca, 2024-11-17:
After going over the mlir-tutorials, I realize that only AIE1 locks are documented. I will take an action item to update them. In the meantime, you can double check your lock logic by studying the output of the objectfifo lowering pass, which you can call with:
`aie-opt --aie-objectFifo-stateful-transform <name of your mlir file>`

ngdymx, 2024-12-05:
Hi, 

Thank you very much! I tried it and although it doesn't work now, I think I can solve it myself. Thanks again!

closed 2024-12-05 by ngdymx