Xilinx/mlir-aie #1745: Compile error(error: expected type): Passthrough_kernel

BoruiXu, 2024-09-05:
Hi,

I ever installed the old version of the NPU driver, and recently, according to this [page](https://github.com/Xilinx/mlir-aie/blob/main/docs/buildHostLin.md), I updated the Linux kernel(6.10) and XDNA Driver(2.18) without reinstallation of the `Xilinx Vitis 2023.2 `. And when using `xrt-smi examine` command, it can show the NPU correctly.  

However, when comping the `passthrough_kernel` in `programing_example`, it encounters the following error.

[image]

I do not know why and how to fix it. Is this because I did not reinstall Xilinx Vitis 2023.2?

Thanks a lot.

keryell, 2024-09-05:
Curious, it looks like the backend LLVM compiler is still using typed pointer and chokes on the new LLVM IR using opaque pointer.
It is equivalent on my laptop to
```
-<%>- /opt/xilinx/Vitis/2023.2/bin/llvm-link --version 
LLVM (http://llvm.org/):
  LLVM version 3.9.0
  Optimized build.
  Default target: x86_64-unknown-linux-gnu
  Host CPU: (unknown)
```
So, this is an LLVM from 2016. Xilinx tools come with a full bag of LLVM versions and for some reasons you are not picking the right one.
Usually, my advice is to never source a Xilinx-provided setup script because it injects so much old stuff that it breaks any normal use later.
Could you try something like
```bash
export XILINX_VITIS=/opt/xilinx/Vitis/2023.2
export XILINX_XRT=/opt/xilinx/xrt
PATH=$PATH:$XILINX_XRT/bin:$XILINX_VITIS/aietools/bin:$XILINX_VITIS/bin
# Put the system libs before XILINX breaking OpenSSL for git
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/lib/x86_64-linux-gnu:/usr/lib:$XILINX_XRT/lib:$XILINX_VITIS/lib/lnx64.o
```
?

BoruiXu, 2024-09-07:
Thanks a lot, however, the compile error still exists.

I do not source the `setup.sh` script provided by the Xilinx and use the following environment variables.
[image]

Do you think reinstall the Xilinx Vitis 2023.2 can fix this error?

closed 2024-09-08 by BoruiXu

keryell, 2024-09-10:
Good you solve the issue. I am curious about what was wrong with the Vitis install.
Lacking AIE support?
Always a pain to install again 100 GB of Vitis software...