Xilinx/mlir-aie #1757: All example kernels execution result in `ERT_CMD_STATE_ERROR` on the Windows host

fxmarty-amd, 2024-09-09:
Hi,

I have been trying to use MLIR-AIE for Ryzen AI NPU on a Windows laptop, with WSL Ubuntu 22.04, without success so far using the examples in https://github.com/Xilinx/mlir-aie/tree/main/programming_examples/basic.

Building the AIE design in WSL works, it is the execution of the resulting `.exe` on Windows side that is failing, specifically the error code from `kernel.wait(...)` is always `4` i.e. `ERT_CMD_STATE_ERROR`, for example in
https://github.com/Xilinx/mlir-aie/blob/fe0c224fb68170d688075902cf8266ead1e1cdcd/programming_examples/basic/passthrough_dmas/test.cpp#L170.

I have tried installing mlir-aie both from the provided wheels or from source (llvm + mlir-aie), none of them being successful. I am using Vitis 2023.2, and used the following install instructions: https://github.com/Xilinx/mlir-aie/blob/main/docs/buildHostWin.md (wheel) & https://github.com/Xilinx/mlir-aie/blob/main/docs/Building.md (from source)

Something that I found doubtful was that running `gendef xrt_coreutil.dll` in `/mnt/c/Technical/xrtNPUfromDLL` gives the logs
```
 * [xrt_coreutil.dll] Found PE+ image
 *** get_primary_data_type '$$Q' unknown
```
but a `xrt_coreutil.def` file is nonetheless created.

On the Windows host side, I use e.g. the following
```
cd buildMSVS
cmake .. -G "Visual Studio 17 2022" -A x64 -DTARGET_NAME=passThroughDMAs
cmake --build . --config Release
.\passThroughDMAs.exe -x ..\build\final.xclbin -i ..\build\insts.txt -k MLIR_AIE -l 4096
```

where printing the result code gives `result code: 5`, and the following check fails. The output matrix is not updated, as if the kernel did not run (which I suppose from the error).

Would you have any idea where the issue may be coming from, or information I could share concerning my setup? cc @stephenneuendorffer @fifield @jgmelber 

Thank you!

fxmarty-amd, 2024-09-13:
Welcome to any suggestion!

Davidqian123, 2025-04-07:
Hello, I have the same issue as you, have you solved the issue?

jgmelber, 2025-04-08:
> Hello, I have the same issue as you, have you solved the issue?

Windows support is currently deprecated. It's likely that it can be hacked to be functional but we are not currently developing or updating Windows support.

Davidqian123, 2025-04-08:
@jgmelber Thanks for your reply, I have switched to Ubuntu 24.04 os. When I try to compile and run [resnet example](https://github.com/Xilinx/mlir-aie/tree/main/programming_examples/ml/resnet/layers_conv2_x,) I faced the following error:
```
python3 /home/dev/Desktop/code/mlir-aie/programming_examples/ml/resnet/layers_conv2_x/test.py -x build/final.xclbin -i build/insts.bin -k MLIR_AIE
Traceback (most recent call last):
  File "/home/dev/Desktop/code/mlir-aie/programming_examples/ml/resnet/layers_conv2_x/test.py", line 489, in <module>
    main(opts)
  File "/home/dev/Desktop/code/mlir-aie/programming_examples/ml/resnet/layers_conv2_x/test.py", line 445, in main
    aie_output = execute(app, ifm_mem_fmt, total_wts3) * block_2_relu_3
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dev/Desktop/code/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/utils/xrt.py", line 262, in execute
    app.run()
  File "/home/dev/Desktop/code/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/utils/xrt.py", line 62, in run
    raise Exception(f"Kernel returned {r}")
Exception: Kernel returned ert_cmd_state.ERT_CMD_STATE_ABORT
make: *** [Makefile:62: run_py] Error 1
```
any suggestions? I have followed README to install and can successfully compile and run [eltwise-add](https://github.com/Xilinx/mlir-aie/tree/main/programming_examples/ml/eltwise_add)

jgmelber, 2025-04-08:
Hi @Davidqian123, which Ryzen AI part are you targeting, specifically the NPU version? 

If you are on Strix or Kracken Point (XDNA2/NPU2) some examples haven't been fully ported #2092.

Davidqian123, 2025-04-08:
I'm using Strix
```
System Configuration
  OS Name              : Linux
  Release              : 6.11.0-21-generic
  Machine              : x86_64
  CPU Cores            : 24
  Memory               : 31394 MB
  Distribution         : Ubuntu 24.04.2 LTS
  GLIBC                : 2.39
  Model                : HP OmniBook Ultra Laptop 14-fd0xxx
  BIOS Vendor          : HP
  BIOS Version         : W81 Ver. 01.01.14

XRT
  Version              : 2.19.0
  Branch               : HEAD
  Hash                 : 21c69cbf48f76980b2bff7be8802414f8c1e17c8
  Hash Date            : 2025-04-08 11:27:02
  amdxdna              : 2.19.0_20250408, deceb066418f2d53fe172a595f59413a731ccee8
  virtio-pci           : unknown, unknown
  NPU Firmware Version : 1.0.7.99

Device(s) Present
|BDF             |Name       |
|----------------|-----------|
|[0000:c4:00.1]  |NPU Strix  |
```

Davidqian123, 2025-04-09:
@jgmelber I'm running [My First Program](https://github.com/Xilinx/mlir-aie/tree/main/programming_guide/section-3) and face the error:
```
python3 /home/dev/Desktop/code/mlir-aie/programming_guide/section-3/test.py -x build/final.xclbin -i build/insts.bin -k MLIR_AIE
[0 0 0 ... 0 0 0]
(4096,)

Error count:  4096

Failed.

make: *** [Makefile:46: run_py] Error 1
```
It seems that the kernel function is not exxecuted correctly, if you could help me take a look at it. If my device should set:
```
from aie.iron.device import NPU1Col1, NPU2
dev = NPU2()
```

jgmelber, 2025-04-09:
The Makefiles attempt to set the correct target device based on an environment variable. For Strix you want to see `export NPU2=1`.

The setup/environment scripts detect the device and set the NPU2 variable: https://github.com/Xilinx/mlir-aie/blob/5474ddd8cf5bd4c958552711f4a2a5de0240e70b/utils/env_setup.sh#L48-L55

jgmelber, 2025-04-10:
> How many examples in [programming_examples](https://github.com/Xilinx/mlir-aie/tree/main/programming_examples) and [programming_guide](https://github.com/Xilinx/mlir-aie/tree/main/programming_guide) can I try using my NPU Strix? Thanks a lot!

Most programming examples have been updated, some remain. This is tracked in #2092. 

As far as I know the programming guide should have all example supported by both architectures. @AndraBisca can you confirm?

Davidqian123, 2025-04-11:
@jgmelber Thanks for updating! I have tried updated examples and I can successfully run `eltwise_mul` and `eltwise_add`, and then I want to combine these two ops to compile as one `final.xclbin` because as I know, we can only pass one xclbin file when testing. However I faced `aie.objectfifo` error when combining these two ops:
```
Traceback (most recent call last):
  File "/home/nexadev/Desktop/code/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/bin/aiecc.py", line 12, in <module>
    main()
  File "/home/nexadev/Desktop/code/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/compiler/aiecc/main.py", line 1351, in main
    run(module_str)
  File "/home/nexadev/Desktop/code/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/compiler/aiecc/main.py", line 1325, in run
    asyncio.run(runner.run_flow())
  File "/usr/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/nexadev/Desktop/code/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/compiler/aiecc/main.py", line 1094, in run_flow
    run_passes(
  File "/home/nexadev/Desktop/code/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/compiler/aiecc/main.py", line 330, in run_passes
    raise e
  File "/home/nexadev/Desktop/code/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/compiler/aiecc/main.py", line 327, in run_passes
    pm.run(module.operation)
aie._mlir_libs._site_initialize.<locals>.MLIRError: Failure while executing pass pipeline:
error: "-":19:5: 'aie.objectfifo' op objectfifo cannot be in more than one ObjectFifoLinkOp
 note: "-":19:5: see current operation: "aie.objectfifo"(%4, %4) <{dimensionsFromStreamPerConsumer = #aie<bd_dim_layout_array_array[[]]>, dimensionsToStream = #aie<bd_dim_layout_array[]>, disable_synchronization = false, elemNumber = 2 : i32, elemType = !aie.objectfifo<memref<2048xbf16>>, plio = false, sym_name = "intBuf", via_DMA = false}> : (index, index) -> ()
error: "-":7:21: 'aie.tile' op number of output DMA channel exceeded!
 note: "-":7:21: see current operation: %4 = "aie.tile"() <{col = 0 : i32, row = 1 : i32}> : () -> index
error: "-":8:26: 'aie.tile' op number of output DMA channel exceeded!
 note: "-":8:26: see current operation: %5 = "aie.tile"() <{col = 0 : i32, row = 0 : i32}> : () -> index
error: "-":7:21: 'aie.tile' op number of output DMA channel exceeded!
 note: "-":7:21: see current operation: %4 = "aie.tile"() <{col = 0 : i32, row = 1 : i32}> : () -> index
error: "-":7:21: 'aie.tile' op number of output DMA channel exceeded!
 note: "-":7:21: see current operation: %4 = "aie.tile"() <{col = 0 : i32, row = 1 : i32}> : () -> index
error: "-":7:21: 'aie.tile' op number of output DMA channel exceeded!
 note: "-":7:21: see current operation: %4 = "aie.tile"() <{col = 0 : i32, row = 1 : i32}> : () -> index
error: "-":7:21: 'aie.tile' op number of input DMA channel exceeded!
 note: "-":7:21: see current operation: %4 = "aie.tile"() <{col = 0 : i32, row = 1 : i32}> : () -> index
error: "-":7:21: 'aie.tile' op number of input DMA channel exceeded!
 note: "-":7:21: see current operation: %4 = "aie.tile"() <{col = 0 : i32, row = 1 : i32}> : () -> index
error: unknown: 'aie.flow' op attribute 'source_channel' failed to satisfy constraint: 32-bit signless integer attribute whose minimum value is 0
 note: unknown: see current operation: "aie.flow"(%4, %3) <{dest_bundle = 1 : i32, dest_channel = 0 : i32, source_bundle = 1 : i32, source_channel = -1 : i32}> : (index, index) -> ()
make: *** [Makefile:51: build/final.xclbin] Error 1
```

Any guidance or suggestion would be appreciated! Here is my code:
https://github.com/Davidqian123/mlir-aie/blob/main/programming_examples/ml/eltwise_mul_add/eltwise_mul_add.py

jgmelber, 2025-04-22:
Somewhat related to your question, you can add two kernels to a single XCLBIN to launch one after another passing intermediate results in DDR as shown in this test: https://github.com/Xilinx/mlir-aie/blob/main/test/npu-xrt/add_one_two/run.lit

Davidqian123, 2025-04-22:
Thanks @jgmelber I will try this solution!

AndraBisca, 2025-04-23:
Thank you for sharing your code @Davidqian123 ! I am currently looking at the last error you signaled and I see two issues:
- one is that the placer allocated resources in such a way that it ran out of Shim tile channels, I am looking into patching this.
- the other is that you are combining a join() and a split() on the same objectfifo `intBuf`, which is not currently supported.

jgmelber, 2025-04-28:
> `'aie.tile' op number of output DMA channel exceeded!`

@Davidqian123 #2221 is a WIP towards fixing a bug you have encountered. 

In the meantime explicitly placing the fill and drain across two shim tiles will work:

https://github.com/Xilinx/mlir-aie/blob/065aae2c805ea4c804b63a466c9df15df429ab08/programming_examples/basic/memcpy/memcpy.py#L93

jgmelber, 2025-05-06:
#2255 for Windows updates

jgmelber, 2025-05-07:
@Davidqian123 

> Any guidance or suggestion would be appreciated! Here is my code:
> https://github.com/Davidqian123/mlir-aie/blob/main/programming_examples/ml/eltwise_mul_add/eltwise_mul_add.py

Here is a patch for the error you were seeing #2221