Xilinx/mlir-aie #1773: Whole Matrix Multiply certain dims produces program memory overflow

gregoryJ-amd, 2024-09-14:
M?=576
K?=768
N?=768
m?=48
k?=64
n?=64
n_aie_cols?=4

For the above dimensions produces the following error. However, larger dimensions i.e. N=1024 does not. 

mkdir -p build
python3 /home/greg/Desktop/mlir-aie/programming_examples/basic/matrix_multiplication/whole_array/aie2.py -m 48 -k 64 -n 64 --n-aie-cols 4 -M 576 -K 768 -N 768 --dtype_in i16 --dtype_out i32 > build/aie_576x768x768_48x64x64_4c.mlir
mkdir -p build
cd build && xchesscc_wrapper aie2 -I /tools/Xilinx/Vitis/2023.2/aietools/include  -DBIT_WIDTH=8 -DDIM_M=48 -DDIM_K=64 -DDIM_N=64 -c /home/greg/Desktop/mlir-aie/programming_examples/basic/matrix_multiplication/whole_array/../../../../aie_kernels/aie2/mm.cc -o mm_48x64x64.o
/home/greg/Desktop/mlir-aie/programming_examples/basic/matrix_multiplication/whole_array/../../../../aie_kernels/aie2/mm.cc:11:9: warning: '__AIENGINE__' macro redefined [-Wmacro-redefined]
#define __AIENGINE__ 2
        ^
<command line>:3:9: note: previous definition is here
#define __AIENGINE__ 1
        ^
In file included from /home/greg/Desktop/mlir-aie/programming_examples/basic/matrix_multiplication/whole_array/../../../../aie_kernels/aie2/mm.cc:23:
In file included from /tools/Xilinx/Vitis/2023.2/aietools/include/aie_api/aie.hpp:10185:
In file included from /tools/Xilinx/Vitis/2023.2/aietools/include/aie_api/aie_adf.hpp:75:
In file included from /tools/Xilinx/Vitis/2023.2/aietools/include/aie_api/adf/stream.hpp:54:
In file included from /tools/Xilinx/Vitis/2023.2/aietools/include/adf.h:7:
/tools/Xilinx/Vitis/2023.2/aietools/include/adf/intrinsics.h:28:9: warning: 'REL_WRITE' macro redefined [-Wmacro-redefined]
#define REL_WRITE -1
        ^
/home/greg/Desktop/mlir-aie/programming_examples/basic/matrix_multiplication/whole_array/../../../../aie_kernels/aie2/mm.cc:20:9: note: previous definition is here
#define REL_WRITE 0
        ^
2 warnings generated.
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <474>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <476>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <478>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <480>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <482>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <484>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <486>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <488>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <474>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <476>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <478>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <480>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <482>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <484>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <486>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_w2c(v16int16, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <488>, would violate structural limitations [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): ... intended destination [-Wkeep-with-operand]
Warning in "": (imprecise line-number, the error occurred somewhere in this function): cannot move keep_with_operand operation `v16acc64 ups_x2c(v16int32, uint6_t, uint1_t, uint1_t, uint2_t, bool &)' <357>, would violate structu [... cut ...]
  File "/home/greg/Desktop/mlir-aie/install/python/aie/compiler/aiecc/main.py", line 1116, in run_flow
    await self.process_cdo()
  File "/home/greg/Desktop/mlir-aie/install/python/aie/compiler/aiecc/main.py", line 561, in process_cdo
    generate_cdo(input_physical.operation, self.tmpdirname)
ValueError: Failed to generate cdo because: 
make: *** [/home/greg/Desktop/mlir-aie/programming_examples/basic/matrix_multiplication/whole_array/../makefile-common:92: build/final_576x768x768_48x64x64_4c.xclbin] Error 1

jgmelber, 2025-04-11:
@gregoryJ-amd is this still an issue?

gregoryJ-amd, 2025-04-28:
No this seems to have since been resolved on the current main latest.

closed 2025-04-28 by gregoryJ-amd