Xilinx/mlir-aie #1793: error: use of undeclared identifier 'hsa_amd_signal_create_on_agent'

thhongee, 2024-09-25:
I tried to install all the toolchain, but I always encounter this error when executing the following command.
Could you please let me know if I missed installing something? 

```
Then, set ${ROCM_ROOT} to the ROCm install. Then, run the following command to build the mlir-aie toolchain targetting the VCK5000 pointing to the externally installed aie-rt and experimental ROCm runtime.

./utils/build-mlir-aie-pcie.sh llvm/build/ build install /opt/xaiengine ${ROCM_ROOT}/lib/cmake/hsa-runtime64/ ${ROCM_ROOT}/lib/cmake/hsakmt/
```

Thanks

Stanley

eddierichter-amd, 2024-09-27:
Hi Stanley,

Thanks for reaching out. I believe you are getting that error because `${ROCM_ROOT}` needs to point to our experimental ROCm runtime that can target the AIEs on the VCK5000 (https://github.com/ROCm/ROCR-Runtime/tree/experimental/rocm-5.6.x-air). 

Have you tried building just with the command `./utils/build-mlir-aie-pcie.sh <llvm dir>/<build dir>`?  That script should build our experimental ROCm runtime and point cmake to it. Let me know if that works. 

Thanks,

Eddie Richter

thhongee, 2024-09-30:
Thanks for your reply,

I got the same error(error: use of undeclared identifier 'hsa_amd_signal_create_on_agent) when I was entering that command:
_./utils/build-mlir-aie-pcie.sh /mnt/cehong/build_vitis_env/mlir-aie/llvm_

Thanks,

Stanley

eddierichter-amd, 2024-09-30:
Hm that is surprising. What commit are you on? Can you see if the following lines are being executed: https://github.com/Xilinx/mlir-aie/blob/main/utils/build-mlir-aie-pcie.sh#L66-L73 in the `build-mlir-aie-pcie.sh` script? This should be building our experimental version of ROCm that has that API defined.

eddierichter-amd, 2024-10-18:
Hi Stanley. Any luck on this?

closed 2025-03-05 by jgmelber