Xilinx/mlir-aie #1833: How to convert AFFine to AIEvecde?

pyl3000, 2024-10-14:
[image]

jsetoain, 2024-10-18:
Can you elaborate a bit, please? Do you have a specific question? Have you tried something and it's not working?

As an aside, I'm not sure where you got the graph you've attached to the issue, but it doesn't look right to me. `mlir-aie` does not produce `aievec` from `affine`, `aievec` is generated from `vector`. `vector` can be generated with upstream tools from `affine` and `linalg`. Also, the C++ generation actually consumes `scf` (and `arith`, `emitc`, etc).

AndraBisca, 2025-03-25:
Closed due to inactivity. Please feel free to re-open when needed.

closed 2025-03-25 by AndraBisca