Xilinx/mlir-aie #1902: How many PLIOs are available from PL to AIE for VCK190 board ?

Albresky, 2024-11-02:
What's the PLIO channels/lanes available at 128-bit for VCK190 board in PL side ? Accourding to UG1079, the PLIOs support 8 connections from PL side to AIEs with 64-bit data width, given that VCK190 has 39 PLIOs, the total connections/channels/lanes at 128-bit from PL to AIEs should be $39 \times (64/128 \times 8)=156$. However, it was weird that **on Vitis2023.2 and later**, aiecompiler says `ERROR: [AIE-PRE-MAPPER-11] area group ((0, 0), (49, 0)) has capacity 78 PLIO 128 bit incoming registered channels`. But on Vitis 2021.1, aiecompiler DO SUPPORT 78 PLIOs and more. Is this a software bug?

FYI, I drafted a discussion on [Xilinx Community](https://adaptivesupport.amd.com/s/question/0D54U00008p7fSPSAY/need-help-maximium-plios-from-pl-to-aie-for-vck190?language=en_US) for details of this issue(including testcases).

Albresky, 2024-11-02:
The issue is raised in this repo because maybe this problem is hardware related, and perhaps MLIR-AIE toolchain could give an answer.

jackl-xilinx, 2024-11-14:
@Albresky I can push the change if you give me permission to push to your branch. For reference, it needs code similar to https://github.com/Xilinx/mlir-aie/blob/main/programming_examples/basic/vector_scalar_mul/run_makefile.lit#L14

jackl-xilinx, 2024-11-15:
Hi @Albresky. You are correct in your calculations. However, at the moment, the tool only allows half the number of incoming channels (78/ 156) to be mapped for 128-bit PLIOs. We're looking into whether accessing the other half can be enabled or not and will update here if there is a workaround.

closed 2024-11-20 by jgmelber

Albresky, 2024-11-21:
> Hi @Albresky. You are correct in your calculations. However, at the moment, the tool only allows half the number of incoming channels (78/ 156) to be mapped for 128-bit PLIOs. We're looking into whether accessing the other half can be enabled or not and will update here if there is a workaround.

Hi, I noticed that this issue has been closed. Has this PLIO problem been solved ?