Xilinx/mlir-aie #1903: Build issue in tutorial2-a

kimm240, 2024-11-04:
When I try to build tutorial2-a, this error occurs.

[image]

When I check directory mlir-aie/install/runtime_lib, There is just 86_64, and aarch64 is not in there.

What should i do to fix that?

kimm240, 2024-11-04:
I follow step: Building on X86 for mlir-aie development in https://github.com/Xilinx/mlir-aie/blob/main/docs/Building.md .

jgmelber, 2024-11-04:
Hi! Which device are you building for?

kimm240, 2024-11-04:
I build it for a Ryzen AI system,
so when I build it,
I use this code
'
source <Vitis Install Path>/settings64.sh
./utils/build-mlir-aie.sh <llvm dir>/<build dir>
'.

And my situation is like this:::
I'm just beginner in mlir so there is no target device,
but just want to check how mlir is compiled so I want to see code is running on x86(cpu).

jgmelber, 2024-11-12:
You shouldn't need aarch64, for this example you should target x86 for the host code and library compilation. Can you remove the aarch64 target from the cmake build command?

jgmelber, 2024-11-12:
The tutorials do not support Ryzen AI, those target embedded AIE devices.

closed 2024-11-18 by jgmelber