Xilinx/mlir-aie #1922: Choose `dimensions` or `sizes` + `strides` for user-facing operations

hunhoffe, 2024-11-15:
Right now, some operations take `dimensions` and some operations take `sizes` and `strides`.

For clarity, but also some implementation considerations, I think it would make sense to change to `sizes`/`strides`.

Some pros for moving to `sizes`/`strides` are:
* `dimensions` requires (at least at the python level) both `sizes`/`strides` to be default or both not default. It would be nice for a user to take advantage of default values more easily. If a python wrapper function takes these values separately, then python needs to have hard-coded defaults in order to produced valid `dimensions`, which is not ideal.
* For `dma_bd`s using the task/chain syntax, each `dma_db` should have at max 3 sizes and at max 4 strides; this is a condition that is not easy to express clearly when using `dimensions`
* Personally, I find it easier to read when they are specified separately (but maybe that's just me)

This came to my attention through comments of @fifield and @andrej in https://github.com/Xilinx/mlir-aie/pull/1919 

As this would be a change that touches many things, I'm recording outside of the PR for discussion and so it does not become a blocker to that PR.

andrej, 2024-11-15:
> For dma_bds using the task/chain syntax, each dma_db should have at max 3 sizes and at max 4 strides; this is a condition that is not easy to express clearly when using dimensions

Why only max 3 sizes but 4 strides? 

Copying my examples from the review of #1919 on why `repeat_count` and the higest-dimension size/wrap aren't the same:

<details>
<summary>Examples</summary>
These examples assume a highest-dimension <b>stride of 2</b>.

If the highest-dimension size/wrap is 3, the higest-dimension stride (=iteration step) is 2, and the repeat count is 6, the highest dimension offset added will be 
<pre>
Task repetition:             0   1   2   3   4   5  (repeat count = 6)
Highest-dim offset added:    0   2   4   0   2   4
                                         ^
                                      wrap (=3)
</pre>

If the highest-dimension wrap matches the repeat count of the task, the iteration step will be added each repetition:

<pre>
Task repetition:             0   1   2   3   4   5  (repeat count = 6)
Highest-dim offset added:    0   2   4   6   8   10 
                                                     ^
                                                 wrap (=6)
</pre>
In the other extreme, if the wrap is 1, the iteration step never gets added (BD repeated with no offset).
<pre>
Task repetition:             0   1   2   3   4   5  (repeat count = 6)
Highest-dim offset added:    0   0   0   0   0   0
                             ^   ^   ^   ^   ^   ^
                             w.  w.  w.  w.  w.  wrap(=1)
</pre>
</details>

hunhoffe, 2026-10-26:
Revisiting this — it's not resolved, and in fact the inconsistency it describes was just reintroduced. aiex.npu.dma_memcpy_nd does use split offsets/sizes/strides (via MyOffsetSizeAndStrideOpInterface), and aie.dma_bd at least prints as split sizes=[...]/strides=[...] even  though its internal storage (BDDimLayoutAttr) is a (size, stride) tuple. But the new aie.objectfifo.dma_endpoint op (added this week by #3964) has a dimensions attribute of combined BDDimLayoutArrayArrayAttr tuples with no pretty-printer, so it prints raw: dimensions = #aie<bd_dim_layout_array_array ...> (see test/objectFifo-split/link_rate_conversion.mlir). The dimensionsToStream [<size=.., stride=..>] surface syntax on aie.objectfifo itself is also still combined-tuple form.

So the decision here was never made uniformly, and the newest code just added another instance of the pattern this issue wants retired. Worth deciding now, before more code depends on ObjectFifoDmaEndpointOp's dimensions field.