Xilinx/mlir-aie #1935: Are there any larger-scale examples that can run on the VCK190?

micropuma, 2024-11-20:
I have completed the basic tutorial for the VCK190, but I’ve noticed that most of the examples in the 'programming_examples' directory are designed for Ryzen AI. I’m wondering if there are any similar examples that I can run on the VCK190 to gain a better understanding of MLIR-AIE? Is there a workflow to lower frontends, such as PyTorch, to backends like MLIR-AIE, enabling the deployment of realistic models?

stephenneuendorffer, 2024-11-20:
We've focused on Ryzen AI as a good, relatively cheap device which allows us to get people up and running relatively easily with AIEngines without getting lost in FPGA programmable logic design.   The VCK190 itself is a relatively limited platform for executing 'realistic models': It's limited in DDR bandwidth, the ARM environment is somewhat constrained for 'big software' development, it requires FPGA development to access more on-device memory, and the datatypes supported by AIE1 are relatively limited compared to state of the art.   The VCK5000 solves some of these problems, enabling work in an X86 host environment and having more (but still constrained) memory bandwidth.  We've worked extensively with both platforms, but both are somewhat challenging to support outside users with (particularly if you're looking for a GPU-like pytorch experience).  I think it's more likely that we'd revisit large models with new AIE-ML based PCIE cards when they come out.

micropuma, 2024-11-20:
Thanks for your reply. I'll pay constant attention.

closed 2024-11-20 by jgmelber