Xilinx/mlir-aie #1954: Error when running whole_array

AJ-RR, 2024-12-03:
Error:

File "mlir-aie/programming_examples/basic/matrix_multiplication/whole_array/aie2.py", line 17, in <module>
    from aie.helpers.taplib import TensorAccessPattern, TensorAccessSequence
ModuleNotFoundError: No module named 'aie.helpers.taplib'

The error seems to occur after fetching recent changes

hunhoffe, 2024-12-06:
Hello! Not sure exactly what is going on, but it seems likely there is something odd going on with Python paths. 

To double check something simple, are you running within the venv through running something like `source ironenv/bin/activate` from the root directory of your clone of mlir-aie?

So I can try to replicate your error, can you also share the steps you took to setup your environment (e.g., did you use the quick setup script or something else?)

AndraBisca, 2025-03-27:
Closed due to inactivity. Please feel free to re-open if needed.

closed 2025-03-27 by AndraBisca