Xilinx/mlir-aie #1964: Some questions about the number of Compute tiles?

ngdymx, 2024-12-10:
Hi team,

I read the tutorial and got that the device is present in HawkPoint (e.g., 8040HS) SOCs. has 5 Columns and 6 Rows, as shown below:

[image]

However, I cannot control the leftmost column CTS. I want to confirm whether we cannot access it. Is there any way to access and control that column? Thanks a lot!

Additionally, the following is noted about device partitions:

[image]

jgmelber, 2024-12-10:
Hi! If you use the device type `npu1` you should be able to  access the whole 5 column device. Remember that you will need to start your shimDMAs at column 1 in that case as there is not a shimTile in the leftmost column.

ngdymx, 2024-12-10:
Hi @jgmelber,

Thanks.

I try it, but get the error shown in the following figure.

[image]

Could you help me with it? Is there any example I can follow? Thanks a lot!

jgmelber, 2024-12-10:
@ngdymx can you share the entire design file, or at least the `@sequence`?

ngdymx, 2024-12-11:
Hi @jgmelber,

Sorry for the late reply. I write a passthrough kernel to test it. It works well when I set the dev is ```AIEDevice.npu1_1col```.  When I change the dev to ```AIEDevice.npu1```, I will get the error I reported yesterday.  I need to rewrite the sequence function but don't know how to do it. Could you help me with it? Thanks!

Case 1:
```python
dev = AIEDevice.npu1_1col
/*************************/
ShimTile = tile(0, 0)
ComputeTile = tile(0, 2)

# To/from AIE-array data movement
@runtime_sequence(tensor_ty, tensor_ty)
def sequence(A, C):
  npu_dma_memcpy_nd(metadata=of_in, bd_id=1, mem=A, sizes=[1, 1, 1, N], issue_token=True)
  npu_dma_memcpy_nd(metadata=of_out, bd_id=0, mem=C, sizes=[1, 1, 1, N], issue_token=True)
  dma_wait(of_in, of_out)
```
The following is the  result: 
[image]

Case 2: 
Then I only change the dev to ```AIEDevice.npu1``` and the corresponding tile.
```python
dev = AIEDevice.npu1
/*************************/
 # Tile declarations
ShimTile = tile(1, 0)
ComputeTile = tile(1, 2)
```
The error I get:
[image]

Here is the project:
Kernel code:
```c
#include <aie_api/aie.hpp>

template <typename T_in, typename T_out>
void passthrough_aie(const T_in *__restrict in0, T_out *__restrict out, const int N) {
    for (int i = 0; i < N; i++) {
            out[i] = in0[i];
    }
}

extern "C" {

    void passThrough(const int32_t *__restrict in0, int32_t *__restrict out, const int N) {
          passthrough_aie<int32_t, int32_t>(in0, out, N);
          
    }
    
}
```
Then the working mlir code: 
```python
import numpy as np

from aie.dialects.aie import *
from aie.dialects.aiex import *
from aie.helpers.dialects.ext.scf import _for as range_
from aie.extras.context import mlir_mod_ctx

import sys

N = 64
dev = AIEDevice.npu1_1col

def pass_loops():
    with mlir_mod_ctx() as ctx:
[... 143 lines cut ...]
  printf("Out:\n");
  for (int i = 0; i < PASSTHROUGH_SIZE; i++) {
    printf("%d", bufOut[i]);
  }

  if (trace_size > 0) {
    test_utils::write_out_trace(((char *)bufOut) +
                                    (PASSTHROUGH_SIZE * sizeof(DATATYPE)),
                                trace_size, vm["trace_file"].as<std::string>());
  }

  // Print Pass/Fail result of our test
  if (!errors) {
    std::cout << std::endl << "PASS!" << std::endl << std::endl;
    return 0;
  } else {
    std::cout << std::endl
              << errors << " mismatches." << std::endl
              << std::endl;
    std::cout << std::endl << "fail." << std::endl << std::endl;
    return 1;
  }
}

```

jgmelber, 2024-12-11:
Thanks for the code, we are looking into it. 

CC @AndraBisca @pvasireddy-amd

AndraBisca, 2024-12-22:
Hello! Just to follow up on this: after some investigating it turned out that there were two related issues here. One [PR](https://github.com/Xilinx/mlir-aie/pull/1997) has been merged into the main branch, the other is well on its way but requires some last cleanups, which will happen after the holidays. Once that [PR](https://github.com/Xilinx/mlir-aie/pull/1971) is in, your design should work!

ngdymx, 2024-12-22:
Hi,

Great, thank you!
Marry Christmas!

victorrjimenezz, 2025-04-02:
Hello!

I'm encountering this same bug. If I try to use the entire NPU --- AIEDevice.npu1 --- I get the `Unsupported tile type at (0, 0) Must be ShimNOC, Mem or Core.` error message. 

I have tried cloning the repo again, pulling the last commit, and running the "env_setup.sh". However, it still does not seem to work.

jgmelber, 2025-04-02:
Hi @victorrjimenezz can you share the code you are using? On NPU1 the tile(0,0) cannot be used for anything, it does not exist in the architecture.

victorrjimenezz, 2025-04-02:
Hi @jgmelber, you are right—I just realized I was using tile(0,0), even though column 0 has no ShimTile. Fixing that resolved the issue. Thanks for the help!