Xilinx/mlir-aie #1967: application kernel call argument limit error

antonio-fc, 2024-12-12:
Hi team,

I'm currently working on the IRON workflow to develop an application for the sake of experimentation. Everything seemed to be working fine until I got to adding the memory buffer arguments to the kernel call as seen in the attached picture ([image]). Up to 5 additional inputs (added to the default input of the function call) have the application working fine, but when attempting to add more, I get a segmentation fault. The memory buffers have been declared and initialized correctly like the others but it still breaks. Is this a feature of the kernel function call or is there something I must do to increase the number of input streams I can use simultaneously? If this is intentional, an explanation is greatly appreciated. The application I'm working on can be found here for reference: https://github.com/antonio-fc/my_mlir-aie/tree/testing/apps/app4

jgmelber, 2024-12-12:
Hi @antonio-fc. I cloned your repo and built `app4` I did not see any errors running the example `test.py`. I ran commit `15329485259b0e413124e65a89c62851b609722c`. My system configuration is: 

```
$ xrt-smi examine
System Configuration
  OS Name              : Linux
  Release              : 6.10.0
  Machine              : x86_64
  CPU Cores            : 16
  Memory               : 62042 MB
  Distribution         : Ubuntu 22.04.3 LTS
  GLIBC                : 2.35
  Model                : F7BSC
  BIOS Vendor          : American Megatrends International, LLC.
  BIOS Version         : 1.04

XRT
  Version              : 2.18.0
  Branch               : HEAD
  Hash                 : 476f42f419bbc5d1545aded3627f03c1c2f1336e
  Hash Date            : 2024-08-13 14:32:08
  xocl                 : unknown, unknown
  xclmgmt              : unknown, unknown
WARNING: xclmgmt version is unknown. Is xclmgmt driver loaded? Or is MSD/MPD running?
  amdxdna              : 2.18.0_20240813, 537a509a3ab1b698c9c9f6ebcd88035b2fe8359b
  NPU Firmware Version : 1.4.2.313

Device(s) Present
|BDF             ||Name          |
|----------------||--------------|
|[0000:c5:00.1]  ||RyzenAI-npu1  |
```

jgmelber, 2024-12-12:
With some experimentation myself using the C++ host code in another design I see this as the max number of arguments: 

```
xrt::kernel(opcode, bo_instr, instr_v.size(), bo_0, bo_1, bo_2, bo_3, bo_4);
```

This is a limitation of the XRT runtime. If you would like more inputs/outputs I suggest concatenating within a `xrt::bo`. You can then use an `offset` within the operations in the `@runtime_sequence`.