Xilinx/mlir-aie #1974: Memory Tile usage

antonio-fc, 2024-12-13:
Hi team,

I am currently trying to use the IRON workflow. I couldn't find information on how to use the Memory Tiles as a way to split data streams between many Compute Tiles, similar to how it can be done with Riallto. Could you point me to where to find information on how to use the Memory Tiles or explain?

AndraBisca, 2024-12-13:
Hello and thank you for your question!

For IRON documentation, the programming guide is a good place to look!

In IRON we typically use the object FIFO data movement primitive as a way to split data from the Mem tiles to Compute tiles. This can take different forms depending on if you are broadcasting the same data to all Compute tiles or distribute parts of the data to each Compute tile:

- Broadcast can be done with an object FIFO that goes from the Mem tile to multiple Compute tiles, where the data arrives in the Memory tile through an input object FIFO and is linked to your output object FIFO. You can find documentation on the broadcast [here](https://github.com/Xilinx/mlir-aie/tree/main/programming_guide/section-2/section-2b/02_Broadcast) and the link [here](https://github.com/Xilinx/mlir-aie/blob/main/programming_guide/section-2/section-2b/03_Link_Distribute_Join/README.md).

- Distribute is achieved via different output object FIFOs that go from the Memory Tile to each Compute Tile, where the data arrives in the Memory tile through an input object FIFO and is linked to your output object FIFOs. Documentation is available [here](https://github.com/Xilinx/mlir-aie/tree/main/programming_guide/section-2/section-2b/03_Link_Distribute_Join#link--distribute).

IRON has also recently been augmented with a higher-level abstraction. We don't have extensive documentation available on it yet, however, all the programming examples use it. Depending on your use case, it might be easier to use.

I hope this helps!

antonio-fc, 2024-12-13:
I see, thanks for the help!

hunhoffe, 2024-12-13:
Hello! 

In addition to the documentation, here is some information on a vector exponentiation programming example that exercises this splitting (and joining) ObjectFifos over a memtile (https://github.com/Xilinx/mlir-aie/tree/main/programming_examples/basic/vector_exp). 

Note that the two designs I like to below are functionally equivalent, just expressed using a different style of IRON.

## Design based on new IRON API
In the newer IRON API, that data is split 

https://github.com/Xilinx/mlir-aie/blob/aff6da32b03c2e236806e90e77eff4f553be054b/programming_examples/basic/vector_exp/vector_exp.py#L39 

and joined

https://github.com/Xilinx/mlir-aie/blob/aff6da32b03c2e236806e90e77eff4f553be054b/programming_examples/basic/vector_exp/vector_exp.py#L42

over a memtile. The default placement of these operations is `AnyMemTile`, so it is not explicitly set in this example code.

## Design based on original IRON API
Using the original style IRON found in the programming guide links above, this operation is supported by calls to `object_fifo_link`:
https://github.com/Xilinx/mlir-aie/blob/aff6da32b03c2e236806e90e77eff4f553be054b/programming_examples/basic/vector_exp/vector_exp_alt.py#L75
https://github.com/Xilinx/mlir-aie/blob/aff6da32b03c2e236806e90e77eff4f553be054b/programming_examples/basic/vector_exp/vector_exp_alt.py#L90

AndraBisca, 2024-12-16:
Just checking whether this issue can be closed? Please let us know if anything is unclear and / or you would like additional information!

closed 2024-12-16 by antonio-fc