Xilinx/mlir-aie #1976: Iterating multiple times over a buffer object

antonio-fc, 2024-12-13:
Hi team,
 I'm trying to implement an application using IRON. My application requires me to multiply the same data array by each element of another data array. However, when trying to iterate over the same memory buffer in a single kernel call, I got a h.wait() error.
Is this something that is possible, and if so, how can it be done?
If more details is needed do let me know. Thanks in advance.

AndraBisca, 2024-12-16:
Hello and thank you for raising this issue! Could you please send us your IRON code?

This application can take several forms depending on the number of compute tiles you would like to use and whether the same data array is small enough to fit in the local memory of a compute tile or needs to be brought it piece by piece into the compute tile(s).

antonio-fc, 2024-12-16:
Hello,

The application can be found at: https://github.com/antonio-fc/my_mlir-aie/tree/testing/apps/app4
In the file 'aie2.py,' I am trying to iterate over the 'u' input twice in the 'runtime_sequence' method using the 'of_in_u' fifo object.
However, despite having the correct input and output size, this fails with a 'h.wait()' error. I found that this happens when I try to iterate over the same input again when I have already iterated over the entire volume once. This is because I want to multiply the entire content more than once. So, I wonder (hope) if this is possible, and if not what is the method to implement this type of operation in a single kernel call?

antonio-fc, 2024-12-16:
in order to run the application, use 'make run_py'

closed 2025-03-04 by antonio-fc