Xilinx/mlir-aie #2006: Add python 3.13 in release

KEKE046, 2025-01-08:
mlir-aie supports only python3.10 and python3.12. Now, python have move to 3.13. python3.13 has fantastic features, such as its JIT compiler, and have been supported by pytorch other common packages. It will be very helpful to build the python3.13 wheels in release.

I checked the workflow file where to add python3.13 support. The following files contain python version info. Maybe we can add `3.13` in `python_versions` to build and upload python3.13 binaries?

https://github.com/Xilinx/mlir-aie/blob/dc34a42a3d82aae3d34d93f4343e540f76f49b9e/.github/workflows/buildRyzenWheels.yml#L143-L149

https://github.com/Xilinx/mlir-aie/blob/dc34a42a3d82aae3d34d93f4343e540f76f49b9e/.github/actions/setup_base/action.yml#L68-L70

stephenneuendorffer, 2025-01-08:
Yes, definitely!

jgmelber, 2025-02-24:
#2063 enables python3.13 wheels for you. FYI our on-board runners will still use python3.10 and python3.12 at this time.