Xilinx/mlir-aie #2011: Issue with README installation instructions

mpardalos, 2025-01-09:
It seems that there is some issue with the installation instructions from the README. In step 4, I get the following output:

```
+ ./build.sh -package
Download npu1 NPUFW version 1.5.1.362:
--2025-01-09 13:41:14--  https://gitlab.com/kernel-firmware/drm-firmware/-/raw/amd-ipu-staging/amdnpu/1502_00/npu.sbin.1.5.1.362
Resolving gitlab.com (gitlab.com)... 2606:4700:90:0:f22e:fbec:5bed:a9b9, 172.65.251.78
Connecting to gitlab.com (gitlab.com)|2606:4700:90:0:f22e:fbec:5bed:a9b9|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2025-01-09 13:41:14 ERROR 404: Not Found.
```

It seems that the link used by build.sh (which I think comes from tools/info.json) is broken. In step 2 the instructions say to use commit 3d5a8cf1af2adfbb6306ad71b45e5f3e1ffc5b37. Are these links maybe fixed in a newer commit?

jgmelber, 2025-01-10:
It is likely the gitlab link for the firmware is out of date. We will need to update our commit hash to a newer version. Feel free to try the latest firmware in the xdna-driver repository

jackl-xilinx, 2025-01-14:
As mentioned by Jeff, those are the new flags needed to build xrt from xdna-driver repo head which worked on our end. Instructions are being updated on https://github.com/Xilinx/mlir-aie/pull/2018

jackl-xilinx, 2025-01-14:
@mpardalos The PR has been merged. Go ahead and follow the updated instructions and see if this works for you.

closed 2025-01-14 by jackl-xilinx