Xilinx/mlir-aie #2017: Inquiring the clock frequency of NPU in the Hawk Point and Strix Point miniPCs

ngdymx, 2025-01-13:
Hi team,

May I ask the clock frequency of the NPU in the Hawk Point and Strix Point miniPCs?

Thanks!

jgmelber, 2025-01-13:
Hawk Point: https://github.com/gyang1099/xdna-driver/blob/96d6e8ab0f2598c439ba76ee8b62ddbf86ed0b45/src/driver/amdxdna/npu1_regs.c#L48-L58
Strix Point: https://github.com/gyang1099/xdna-driver/blob/96d6e8ab0f2598c439ba76ee8b62ddbf86ed0b45/src/driver/amdxdna/npu4_regs.c#L20-L30

closed 2025-01-13 by jgmelber

newling, 2025-03-19:
@jgmelber  I was wondering the same thing wrt Hawk Point and came across this question, what is the answer? i.e. how should the table be interpreted? 

```
const struct dpm_clk_freq npu1_dpm_clk_table[] = {
	{400, 800},
	{600, 1024},
	{600, 1024},
	{600, 1024},
	{600, 1024},
	{720, 1309},
	{720, 1309},
	{847, 1600},
	{ 0 }
};
```

ngdymx, 2025-03-20:
Hi @newling,

I am working on the [Hawk Point Mini PC: GMKtec NucBox K8 : AMD Ryzen™ 7 8845HS](https://www.gmktec.com/products/amd-ryzen-7-8845hs-mini-pc-nucbox-k8?spm=..product_fe40bedf-d378-40fc-a60a-54d18f1dbc53.header_1.1&variant=71118fa4-6acb-4d6e-abcb-1d3cf00e6438).

Here is the information I got, hope it is helpful.

[image]

[image]

jgmelber, 2025-03-20:
Those screenshots are helpful. The larger number should be the clock frequency for the AIE array. The smaller number as you can see in your screenshot is for the MP-IPU also known as the dedicated control processor that interprets the runtime sequence operations.

newling, 2025-03-20:
Ok. Thanks for the screenshots @ngdymx. I guess then that with 16 NPU cores running at 1.6 GHz, in theory it should be capable of 128 MACs (bf16) * 2 ('*' and '+' in flop) * 16 (cores) * 1.6e9 (clock) = 6.4 tflops? 

(FYI @jtuyls)

jgmelber, 2025-11-20:
Strix, Strix Halo, Krackan "NPU4" table: 

https://github.com/amd/xdna-driver/blob/e8c3646f84050d48b881fb9850f4f4868ede98ee/src/driver/amdxdna/npu4_regs.c#L40-L50