Xilinx/mlir-aie #2045: Add python bindings for stack size in core definition

jackl-xilinx, 2025-02-14:
Currently in mlir, we can pass in a parameter to the core object to specify stack size as follows:

```
AIE.core(%tile) {
      AIE.end
} { stackSize = 2048 : i32, elf_file = "core_33.elf" }
```

But this option doesn't exist for the python bindings. This should be added.

closed 2025-03-07 by AndraBisca

elliottbinder, 2025-03-11:
I've tried this out, and I am able to compile programs using the previously unavailable local memory (i.e. by setting stack_size=0), but some functions are still using the stack. Is there a way the compiler can give feedback when the requested stack size is not respected?

fifield, 2025-03-11:
> I've tried this out, and I am able to compile programs using the previously unavailable local memory (i.e. by setting stack_size=0), but some functions are still using the stack. Is there a way the compiler can give feedback when the requested stack size is not respected?

This is certainly possible but there is currently no support for this feedback in mlir-aie. There has been some related work recently in the iree-amd-aie repo which might be of interest. See for example https://github.com/nod-ai/iree-amd-aie/pull/1121. mlir-aie could integrate similar functionality to provide feedback when the requested stack size cannot be accommodated by the single core compiler.

PR #2106 (2025-03-07): Add worker stack size to python bindings
Closes #2045

Note that the order of the positional arguments does not correspond to the order used in mlir. I did not change this to avoid breaking all previous tests using positional arguments when declaring a core.