Xilinx/mlir-aie #2092: NPU Programming Examples

jgmelber, 2025-03-04:
Work started in #1986.

vision:
- [ ] fix color_detect

basic:
- [ ] fix tiling_exploration 
- [ ] fix vec_exp
- [ ] port matrix_multiplication

ml:
- [ ] fix conv2d
- [ ] fix conv2d_fused_relu
- [ ] port bottleneck
- [ ] port resnet

jgmelber, 2025-03-04:
Add an environment variable to set NPU2 in Makefiles

jgmelber, 2025-03-05:
> * [ ]  Add an environment variable to set NPU2 in Makefiles

#2103

AndraBisca, 2025-03-28:
vector_exp updated by #2163

jgmelber, 2025-04-08:
#2189 helps any LUT kernels for AIE2P

jgmelber, 2025-04-10:
Many of these examples need single core kernels for aie2p in addition to the currently available aie2 versions. This requires slight adaptations specific to the architecture still using the [aie_api](https://xilinx.github.io/aie_api/) C++ library.  The kernels are generally located here: https://github.com/Xilinx/mlir-aie/tree/main/aie_kernels

AndraBisca, 2025-06-02:
#2345 

[x] fix bottleneck
[x] fix resnet

closed 2026-05-29 by jgmelber