Xilinx/mlir-aie #2125: Error: (E109) complete binding failed during aie simulation on programming_examples

Albresky, 2025-03-12:
Hi, I've encountered an error when simulating the programming example **_mlir-aie/programming_examples/mlir/MM_2x2/circuit_switched_version_** from mlir-aie project.

I've successufully compiled this example with `--aiesim` option and got the below file structure:

```bash
.
├── aie.mlir
├── aie.mlir.prj
│   ├── aie_inc.cpp
│   ├── aiesim.sh
│   ├── core_6_3.bcf
│   ├── core_6_4.bcf
│   ├── core_7_3.bcf
│   ├── core_7_4.bcf
│   ├── input.ll
│   ├── input.llchesshack.ll
│   ├── input.llchesslinked.ll
│   ├── input.o
│   ├── input.o.lst
│   ├── input_opt_with_addresses.mlir
│   ├── input_physical.mlir
│   ├── input_with_addresses.mlir
│   ├── sim
│   │   ├── arch
│   │   │   └── aieshim_solution.aiesol
│   │   ├── config
│   │   │   └── scsim_config.json
│   │   ├── flows_physical.json
│   │   ├── flows_physical.mlir
│   │   ├── ps
│   │   │   └── ps.so
│   │   └── reports
│   │       └── graph.xpe
│   └── work
│       ├── input.llchesslinked.aliases
│       ├── input.llchesslinked.ctt
│       ├── input.llchesslinked.dti
│       ├── input.llchesslinked-Fcore_6_3_.o
│       ├── input.llchesslinked-Fcore_6_3_.sfg
│       ├── input.llchesslinked-Fcore_6_4_.o
│       ├── input.llchesslinked-Fcore_6_4_.sfg
│       ├── input.llchesslinked-Fcore_7_3_.o
│       ├── input.llchesslinked-Fcore_7_3_.sfg
│       ├── input.llchesslinked-Fcore_7_4_.o
│       ├── input.llchesslinked-Fcore_7_4_.sfg
│       ├── input.llchesslinked-Fllvm___aie___lock___acquire___reg_.o
│       ├── input.llchesslinked-Fllvm___aie___lock___acquire___reg_.sfg
│       ├── input.llchesslinked-Fllvm___aie___lock___release___reg_.o
│       ├── input.llchesslinked-Fllvm___aie___lock___release___reg_.sfg
│       ├── input.llchesslinked.fnm
│       ├── input.llchesslinked.gvt
│       ├── input.llchesslinked.gvt.o
│       ├── input.llchesslinked.ini
│       ├── input.llchesslinked.lib
│       ├── input.llchesslinked.objlist
│       └── input.llchesslinked.sfg
├── AIESimulator.log
├── aiesimulator_output
│   ├── aiesim_options.txt
│   └── default.aierun_summary
├── core_6_3.elf
├── core_6_3.elf.map
├── core_6_4.elf
├── core_6_4.elf.map
├── core_7_3.elf
├── core_7_3.elf.map
├── core_7_4.elf
├── core_7_4.elf.map
├── diag_report.log
├── elf
[... 778 lines cut ...]
    aie.wire(%tile_9_1 : Core, %switchbox_9_1 : Core)
    aie.wire(%tile_9_1 : DMA, %switchbox_9_1 : DMA)
    aie.wire(%switchbox_8_2 : East, %switchbox_9_2 : West)
    aie.wire(%tile_9_2 : Core, %switchbox_9_2 : Core)
    aie.wire(%tile_9_2 : DMA, %switchbox_9_2 : DMA)
    aie.wire(%switchbox_9_1 : North, %switchbox_9_2 : South)
    aie.wire(%shim_mux_10_0 : North, %switchbox_10_0 : South)
    aie.wire(%shim_noc_tile_10_0 : DMA, %shim_mux_10_0 : DMA)
    aie.wire(%switchbox_9_1 : East, %switchbox_10_1 : West)
    aie.wire(%tile_10_1 : Core, %switchbox_10_1 : Core)
    aie.wire(%tile_10_1 : DMA, %switchbox_10_1 : DMA)
    aie.wire(%switchbox_10_0 : North, %switchbox_10_1 : South)
    aie.flow(%shim_noc_tile_6_0, DMA : 0, %tile_7_3, DMA : 0)
    aie.flow(%shim_noc_tile_6_0, DMA : 0, %tile_6_3, DMA : 0)
    aie.flow(%shim_noc_tile_6_0, DMA : 1, %tile_7_4, DMA : 0)
    aie.flow(%shim_noc_tile_6_0, DMA : 1, %tile_6_4, DMA : 0)
    aie.flow(%tile_6_4, DMA : 0, %shim_noc_tile_6_0, DMA : 0)
    aie.flow(%shim_noc_tile_7_0, DMA : 0, %tile_6_3, DMA : 1)
    aie.flow(%shim_noc_tile_7_0, DMA : 1, %tile_6_4, DMA : 1)
    aie.flow(%tile_7_4, DMA : 0, %shim_noc_tile_6_0, DMA : 1)
    aie.flow(%shim_noc_tile_10_0, DMA : 0, %tile_7_3, DMA : 1)
    aie.flow(%shim_noc_tile_10_0, DMA : 1, %tile_7_4, DMA : 1)
  }
}
```

Albresky, 2025-03-13:
Hi @jackl-xilinx, 
Is this a tool-related problem? I've tested those cases under `MM_2x2` example, they all failed with the same error message(_"2 binds exceeds maximum of 1 allowed"_). I've checked these aie.mlir files again and I think the IR is correct.

jgmelber, 2025-03-25:
We don't currently support simulation for NPU designs

closed 2025-03-25 by jgmelber

Albresky, 2025-03-25:
> We don't currently support simulation for NPU designs

I see, but I've targeted the device into xcvc1902. I think the aie-simulator should work.