Xilinx/mlir-aie #2129: undef_v8acc32() meaning

flxinxout, 2025-03-14:
Dear developers,

I want to ask the meaning of all these `undef_` functions for data types in the AIE-API, especially if it has context in the low-level architecture of AIE-ML.
Im struggling to use the FFTs because of many errors of the form

```
Workersld.lld: error: undefined symbol: broadcast_zero_to_v8cacc64()
>>> referenced by dft_integer.cc
>>>               dft_integer.o:(dft_basic_integer)

ld.lld: error: undefined symbol: undef_v8cacc64()
>>> referenced by dft_integer.cc
>>>               dft_integer.o:(dft_basic_integer)
clang: error: ld.lld command failed with exit code 1 (use -v to see invocation)
Error encountered while running: /home/sens/mlir-aie/ironenv/lib/python3.12/site-packages/llvm-aie/bin/clang -O2 --target=aie2-none-elf /home/sens/mlir-aie/programming_examples/basic/dft_integer/build/aie.mlir.prj/input.o -Wl,--gc-sections -Wl,--orphan-handling=error -Wl,-T,/home/sens/mlir-aie/programming_examples/basic/dft_integer/build/aie.mlir.prj/core_0_2.ld.script -o ./core_0_2.elf

```

stephenneuendorffer, 2025-03-17:
The "undef" methods are a small optimization, which enables vector types to be defined without implicit initialization, which can sometimes save a cycle or two in an inner loop.  Can you share the code you're actually trying to compile?  I experimented a bit, also looking at the issue you created here: https://github.com/Xilinx/aie_api/issues/3 and was unable to replicate what you're seeing.   Another possibility would be to install Vitis separately and use the chess single core compiler (something like `aiecc.py --xchesscc --xbridge`)

flxinxout, 2025-03-19:
dear @stephenneuendorffer thanks for you reply

My actual code is the following: I created a kernel file that uses the example of 128 point FFT from AIE API. Ignore the the input to the function `dft_basic` because right now it does not matter what I input as the function just run the example.

```
void test_fft(){
    constexpr unsigned        n = 128;
    constexpr unsigned shift_tw = 15;
    constexpr unsigned    shift = 15;
    constexpr bool          inv = false;
    
    alignas(aie::vector_decl_align) static cint32_t x[n];
    alignas(aie::vector_decl_align) static cint32_t tmp[n];
    alignas(aie::vector_decl_align) static cint32_t y[n];
    
    alignas(aie::vector_decl_align) static cint16_t tw1    [] = {{ 32767,      0}};
    alignas(aie::vector_decl_align) static cint16_t tw2    [] = {{ 32767,      0}, {     0, -32768}};
    alignas(aie::vector_decl_align) static cint16_t tw4    [] = {{ 32767,      0}, { 23170, -23170}};
    alignas(aie::vector_decl_align) static cint16_t tw2_4  [] = {{ 32767,      0}, {-23170, -23170}};
    alignas(aie::vector_decl_align) static cint16_t tw8    [] = {{ 32767,      0}, { 30273, -12539}, { 23170, -23170}, { 12539, -30273},
                                                            {     0, -32768}, {-12539, -30273}, {-23170, -23170}, {-30273, -12539}};
    alignas(aie::vector_decl_align) static cint16_t tw16   [] = {{ 32767,      0}, { 32138,  -6392}, { 30273, -12539}, { 27245, -18204},
                                                            { 23170, -23170}, { 18204, -27245}, { 12539, -30273}, {  6392, -32138}};
    alignas(aie::vector_decl_align) static cint16_t tw8_16 [] = {{ 32767,      0}, { 27245, -18204}, { 12539, -30273}, { -6392, -32138},
                                                            {-23170, -23170}, {-32138,  -6392}, {-30273,  12539}, {-18204,  27245}};
    alignas(aie::vector_decl_align) static cint16_t tw32   [] = {{ 32767,      0}, { 32610,  -3211}, { 32138,  -6392}, { 31357,  -9512},
                                                            { 30273, -12539}, { 28898, -15446}, { 27245, -18204}, { 25330, -20787},
                                                            { 23170, -23170}, { 20787, -25330}, { 18204, -27245}, { 15446, -28898},
                                                            { 12539, -30273}, {  9512, -31357}, {  6392, -32138}, {  3211, -32610},
                                                            {     0, -32768}, { -3211, -32610}, { -6392, -32138}, { -9512, -31357},
                                                            {-12539, -30273}, {-15446, -28898}, {-18204, -27245}, {-20787, -25330},
                                                            {-23170, -23170}, {-25330, -20787}, {-27245, -18204}, {-28898, -15446},
                                                            {-30273, -12539}, {-31357,  -9512}, {-32138,  -6392}, {-32610,  -3211}};
    alignas(aie::vector_decl_align) static cint16_t tw64   [] = {{ 32767,      0}, { 32728,  -1607}, { 32610,  -3211}, { 32413,  -4808},
                                                            { 32138,  -6392}, { 31785,  -7961}, { 31357,  -9512}, { 30852, -11039},
                                                            { 30273, -12539}, { 29621, -14010}, { 28898, -15446}, { 28106, -16846},
                                                            { 27245, -18204}, { 26319, -19519}, { 25330, -20787}, { 24279, -22005},
                                                            { 23170, -23170}, { 22005, -24279}, { 20787, -25330}, { 19519, -26319},
                                                            { 18204, -27245}, { 16846, -28106}, { 15446, -28898}, { 14010, -29621},
                                                            { 12539, -30273}, { 11039, -30852}, {  9512, -31357}, {  7961, -31785},
                                                            {  6392, -32138}, {  4808, -32413}, {  3211, -32610}, {  1607, -32728}};
    alignas(aie::vector_decl_align) static cint16_t tw32_64[] = {{ 32767,      0}, { 32413,  -4808}, { 31357,  -9512}, { 29621, -14010},
                                                            { 27245, -18204}, { 24279, -22005}, { 20787, -25330}, { 16846, -28106},
                                                            { 12539, -30273}, {  7961, -31785}, {  3211, -32610}, { -1607, -32728},
                                                            { -6392, -32138}, {-11039, -30852}, {-15446, -28898}, {-19519, -26319},
                                                            {-23170, -23170}, {-26319, -19519}, {-28898, -15446}, {-30852, -11039},
                                                            {-32138,  -6392}, {-32728,  -1607}, {-32610,   3211}, {-31785,   7961},
                                                            {-30273,  12539}, {-28106,  16846}, {-25330,  20787}, {-22005,  24279},
                                                            {-18204,  27245}, {-14010,  29621}, { -9512,  31357}, { -4808,  32413}};
    
    // Constant value input
    for (unsigned i = 0; i < n; ++i) {
        x[i] = cint32_t(1, 0);
    }
    
    aie::fft_dit_r2_stage<64, cint32_t, cint32_t, cint16_t>(x,   tw1,                   n, shift_tw, shift, inv, tmp);
    aie::fft_dit_r4_stage<16, cint32_t, cint32_t, cint16_t>(tmp, tw2,   tw4,   tw2_4,   n, shift_tw, shift, inv, y);
[... cut ...]
	${powershell} ./$< -x build/final.xclbin -i build/insts.txt -k MLIR_AIE

trace: ${targetname}.exe build/final_trace.xclbin build/insts.txt 
	${powershell} ./$< -x build/final_trace.xclbin -i build/insts.txt -k MLIR_AIE
	${srcdir}/../../utils/parse_trace.py --filename trace.txt --mlir build/aie_trace.mlir --colshift 1 > trace_vs.json
	${srcdir}/../../utils/get_trace_summary.py --filename trace_vs.json

clean_trace:
	rm -rf tmpTrace trace.txt parse*json trace*json

clean: clean_trace
	rm -rf build _build ${targetname}*.exe

```

The errors I get are always:

[image]

flxinxout, 2025-03-20:
Dear Developers:

As answered finally in https://github.com/Xilinx/aie_api/issues/3 the need of Vitis tools is required to use FFTs of AIE-API, as the LLVM-AIE compiler is not supporting right now these intrinsics.