Xilinx/mlir-aie #2131: Failing Run Examples on Ryzen AI (amd7940hs)

AndraBisca, 2025-03-14:
Noticed this on #2066 where programming_examples/ml/relu/ failed at trace generation. I commented out the lines in the Makefile to bypass this error but it should be possible to add them back.

jackl-xilinx, 2025-03-19:
Testing shows that running with trace enabled (8192 trace buffer size) seg faults. Adding a -v 1 to the executable command line made it pass intermittently (seg fault at other times). It's unclear why this argument makes a difference. We have compared the `test.cpp`, `Makefile`, and `CMakeList.txt` to `passthrough_kernel` which is similar data access pattern to this design and that works without issue. Investigation ongoing and trace testing will not be added back to `run_makefile_placed.lit` until it is figured out.

jackl-xilinx, 2025-04-10:
This issue was fixed with the new refactored test.cpp/xrt_test_wrapper.h where defining unused XRT buffers causes the seg faults to stop occuring. See https://github.com/Xilinx/mlir-aie/pull/2195

closed 2025-04-10 by jackl-xilinx