Xilinx/mlir-aie #2135: Enabling trace for matmul single_core causes intermittent functional failure when run

jackl-xilinx, 2025-03-14:
Without trace, matmul single_core seems to pass functional test consistently. When trace is added, it seems to fail intermittently which is the reason `make trace` was not added to the `run_makefile.lit`. This should be investigated and fixed so this trace is tested for matmul.

jackl-xilinx, 2025-05-12:
Current workaround is to ensure trace buffer is much larger than allocated in dma since seg fault seems to be related to host memory being overwritten by dma.

fifield, 2025-05-13:
> Current workaround is to ensure trace buffer is much larger (4x) than allocated in dma since seg fault seems to be related to host memory being overwritten by dma.

Can't the trace dma transfer size be limited to the buffer size? I don't quite understand why you can't eliminate the potential for overflow.

jackl-xilinx, 2025-05-14:
> Can't the trace dma transfer size be limited to the buffer size? I don't quite understand why you can't eliminate the potential for overflow.

It doesn't seem to be trace that's overwriting the buffer per se but something else in the driver that's doing it. The dma was originally sized to be the same size as the buffer but that led to intermittent overflow that's currently pinned to a driver issue but needs further investigation. This doesn't occur in windows because the buffer memory is protected but on linux, that's not the case.

fifield, 2025-05-14:
> It doesn't seem to be trace that's overwriting the buffer per se but something else in the driver that's doing it. The dma was originally sized to be the same size as the buffer but that led to [...]

Thanks for the explanation. It's odd because there's nothing different about that buffer vs. any other data buffer, as far as the driver is concerned.

joeldushouyu, 2025-06-06:
> [@jackl-xilinx](https://github.com/jackl-xilinx) and [@fifield](https://github.com/fifield)
> 
> Recently, we encountered a bug with trace(unfortunately, it is a different bug having to do with data racing; I will raise another issue for that). 

I just realized the other issue I was referring to above is actually caused by the same //4 issue. 

I can create a PR to fix, probably by doing a divide by 4 for all  python calls  configure_packet_tracing_aie2()?
Or should this detail be hidden from user and do it inside the configure_packet_tracing_aie2() function instead?