Xilinx/mlir-aie #2136: Scalar version of matrix_multiplication single_core fails when M, K, N is specified to 1024

JinmingZhuang, 2025-03-16:
Hi all,

I encountered with a runtime error multiple times shown below when calling host executable: 

```sh
terminate called after throwing an instance of 'std::runtime_error'
  what():  qds_device::wait() unexpected command state
Aborted (core dumped)
```

I guess this issue may be related to the ID assignment and data synchronization mechanism (```aiex.npu.dma_wait```) in ```aiex.runtime_sequence```. From my observations, the error consistently appears when a certain number of ```dma_memcpy_nd``` operations are present. The corresponding generated MLIR code is attached for your reference.

[aie_1024x1024x1024_64x64x64.txt](https://github.com/user-attachments/files/19275697/aie_1024x1024x1024_64x64x64.txt)

This issue can be reproduced, when using the scalar version of matrix multiplication for large matrix size. I set the size to M=1024, K=1024, N=1024 in the Makefile and set the 90th line of ```single_core.py``` to ```vectorized = False```.

I would appreciate any insights or clarifications regarding the requirements for ```dma_memcpy_nd``` within ```aiex.runtime_sequence```.

Thank you very much for your help in advance!

Best,

jgmelber, 2025-03-20:
> I would appreciate any insights or clarifications regarding the requirements for dma_memcpy_nd within aiex.runtime_sequence.

https://github.com/Xilinx/mlir-aie/blob/main/programming_guide/section-2/section-2d/DMATasks.md#best-practices-for-data-movement-and-synchronization-with-npu_dma_memcpy_nd

Please let me know if that guide helps and if there is anything you would like additional clarity on.

JinmingZhuang, 2025-03-20:
Thank you very much for providing the useful information! I will find more information in the programming guide.

closed 2025-03-20 by JinmingZhuang