Xilinx/mlir-aie #2140: Invalid routes over undeclared tiles

jackl-xilinx, 2025-03-19:
There is a bug right now if routes from packet flows (for trace) go over tiles that are not declared in the design. When that occurs, the switchbox declarations for those routes are not elaborated in the physical mlir. This can be worked around by declaring all tiles or at least those tiles where routes are likely to be routed (bounding box for the entire design) but this should be generally without needing this workaround.

jackl-xilinx, 2025-03-20:
Declaring tiles in python looks something like this:

```
DummyTile01 = tile (0,1)
DummyTile02 = tile (0,2)
DummyTile03 = tile (0,3)
DummyTile04 = tile (0,4)
DummyTile10 = tile (1,0)
DummyTile11 = tile (1,1)
```

erwei-xilinx, 2025-03-23:
This PR should fix the issue.

jgmelber, 2025-03-29:
@erwei-xilinx can this issue be closed?

jackl-xilinx, 2025-04-03:
Verified fix on example test failure.