Xilinx/mlir-aie #2150: Section 3b - hotfix

GiuseppeSorrentino99, 2025-03-22:
It is just a minor issue, but I could not run the section3 test.py as the following lines: 
mlir-aie/programming-guide/section-3/test.py:19 

```sh
with open(opts.instr, "r") as f:
        instr_text = f.read().split("\n")
        instr_text = [l for l in instr_text if l != ""]
        instr_v = np.array([int(i, 16) for i in instr_text], dtype=np.uint32)
```

fail to read the intr.txt file.

In my case, I fixed with 

```
with open(opts.instr, "rb") as f:
    content = f.read()
    instr_v = np.frombuffer(content, dtype=np.uint32)
```
and now the test regularly pass

jgmelber, 2025-03-22:
Are you using the latest tools? There was a recent update #2128 that may be impacting your flow. The default now is to generate the instruction sequence as a binary rather than text file. The Python helpers will select text or binary parsing based on the extension of your file. Perhaps your file is misnamed for the format the tool is generating. 

https://github.com/Xilinx/mlir-aie/blob/51c7549c56b8f28551ba30b831b2b8009af93fb2/python/utils/xrt.py#L162-L176

GiuseppeSorrentino99, 2025-03-22:
Thanks for answering. Indeed the recent update you pointed me at solved the issue. Thanks!

closed 2025-03-22 by GiuseppeSorrentino99