Xilinx/mlir-aie #2180: Conv2d test fails checks in Strix

mariodruiz, 2025-04-03:
I am running this example https://github.com/Xilinx/mlir-aie/tree/main/programming_examples/ml/conv2d in both Phoenix and Strix manchines.

In Phoenix the test passes and the absolute max and min error is:

```python
0.007800006055831921
1.4781951773734647e-09
```

In Strix the test fails and the absolute max and min error is:

```python
0.3510000026702881
1.4781951773734647e-09
```

Also the https://github.com/Xilinx/mlir-aie/tree/main/programming_examples/ml/conv2d_fused_relu for Strix is giving this error when doing `make run_py`

```python
python3  /notebooks/mlir-aie/programming_examples/ml/conv2d_fused_relu/test.py -x build/final.xclbin -i build/insts.bin -k MLIR_AIE
Traceback (most recent call last):
  File "/notebooks/mlir-aie/programming_examples/ml/conv2d_fused_relu/test.py", line 167, in <module>
    main(opts)
  File "/notebooks/mlir-aie/programming_examples/ml/conv2d_fused_relu/test.py", line 123, in main
    aie_output = execute(app, ifm_mem_fmt, total_wts) * relu_scale
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mliraie/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/utils/xrt.py", line 262, in execute
    app.run()
  File "/home/mliraie/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/utils/xrt.py", line 60, in run
    r = h.wait()
        ^^^^^^^^
RuntimeError: qds_device::wait() unexpected command state
```

AndraBisca, 2025-04-03:
Hello Mario! I think this is related to #2092 which tracks a few examples which were not yet ported to Strix, conv2d_fused_relu among them. I am currently working on porting them, I will update you when I have a working version!

closed 2025-04-03 by AndraBisca

mariodruiz, 2025-04-03:
Thank you Andra