Xilinx/mlir-aie #2186: Potential issue with aiex.npu.dma_memcpy_nd lowering

erwei-xilinx, 2025-04-03:
In this matrix multiplication example: https://gist.github.com/erwei-xilinx/732031161bc0e77823ba14a7767a34ca

Two runtime sequences were implemented, which I think are equivalent. The first one fails with driver timeout. The second one passes.

The failing one has 

```
aiex.npu.dma_memcpy_nd(%arg1[0, 0, 0, 0][32, 2, 64, 64][16384, 64, 256, 1]) {id = 0 : i64, metadata = @airMemcpyId10} : memref<4096x256xbf16>
aiex.npu.dma_memcpy_nd(%arg1[0, 0, 0, 524288][32, 2, 64, 64][16384, 64, 256, 1]) {id = 1 : i64, metadata = @airMemcpyId10} : memref<4096x256xbf16>
```

The passing one has

```
aiex.npu.dma_memcpy_nd(%arg1[0, 0, 0, 0][64, 2, 64, 64][16384, 64, 256, 1]) {id = 0 : i64, metadata = @airMemcpyId10} : memref<4096x256xbf16>
```
instead.

Could that indicate a bug, or am I missing something?

erwei-xilinx, 2025-04-03:
Cc: @fifield , @jgmelber , @jackl-xilinx , @endtaka-amd , @AndraBisca .

jgmelber, 2025-04-03:
I can't see why one would fail versus the other. If anything I would think a transfer with a higher number in one index would fail due to some out of bounds...

erwei-xilinx, 2025-04-03:
> I can't see why one would fail versus the other. If anything I would think a transfer with a higher number in one index would fail due to some out of bounds...

Yeah both seem equivalent to me, and I think both are within the wrap bound of the hardware, where the max wrap for left-most dimension is 64.

hunhoffe, 2026-11-03:
I tried this again on current main with a minimal standalone repro. The shim reads memref<4096x256xbf16> through a memtile passthrough, using exactly the two runtime sequences above, and writes the result to a contiguous buffer, which is checked element by element against a numpy gather with the same sizes and strides.

On npu2 (Strix Halo) both variants complete and match exactly: 0 / 524288 mismatches, no timeout. The lowering is also as expected on npu1 and npu2. The split version emits two BDs on the same channel that are identical except for the address patch: the second is at +1048576 bytes (524288 elements = 32 × 16384, the offset row), and each queue push uses repeat_count = 31. So the large innermost offset is folded into the base address correctly.                                                                            

@erwei-xilinx, does the original AIR design still time out for you on current main? If it does, could you attach the full MLIR? Otherwise I'd suggest we add this pattern as an npu-xrt regression test and close the issue.