Xilinx/mlir-aie #2187: Data Corruption When Using Multiple bd_ids with npu_dma_memcpy_nd

victorrjimenezz, 2025-04-03:
Hello!

I am encountering an issue when transferring data from the same Shim to the same object FIFO with different `bd_id`. The code is the following:

        for iter in range(n_parallel_memcpy):
            npu_dma_memcpy_nd(
            metadata=object_fifo_on_shim_1_0,
            bd_id=iter,
            mem=A,
            offsets=[0, 0, 0, iter * size],
            sizes=[1, 1, 1, size],
            strides=[0, 0, 0, 1],)

_Iff_ `n_parallel_memcpy` > 1, I observe corrupted data — 1 element in a 64x64 matrix is corrupted. This causes my execution to output incorrect results.
I suspect that I may not be using `npu_dma_memcpy_nd` properly. Specifically, I am interested in understanding:

- Does the ObjectFifo receive the data in the order that the `npu_dma_memcpy_nd` is executed in Python?
- Could the order be non-deterministic or undefined, requiring synchronization before issuing another `npu_dma_memcpy_nd` on the same ObjectFifo?

If synchronization is necessary, I am unsure why there are 16 `bd_ids` available while the same ShimTile has only 3 channels.

victorrjimenezz, 2025-04-04:
Also, I have tried replacing `npu_dma_memcpy_nd` with 

            bd = 2
            for i in range(6):
                for col in range(4):
                    tasks.append(shim_dma_single_bd_task(
                        alloc=shim_out_vector[col],
                        mem=C,
                        offset=col * M + i * m * 4,
                        sizes=[1, 1, 1, m * 4],
                        strides=[0, 0, 0, 1],
                        issue_token=True
                    ))
                    dma_start_task(tasks[-1])

                bd += 1
                if bd == 4:
                    for col in range(4):
                        for task in vector_out_tasks:
                            dma_await_task(task)
                        bd = 0
                        vector_out_tasks = []

Although I get a lot more correct elements, the results are still 0 after a few iterations of i.

fifield, 2025-04-07:
> The first iteration of i seems to be correct, and the output seems to be 0 at the second iteration of i. However, if I perform a `dma_wait` for every iteration of the `for i range(6)` the code works.

For a transfer from the shim to host memory, the `issue_token` attribute of the `npu_dma_memcpy_nd` op defaults to `true`. That means that each of the tasks (memcopies) enqueued to the shim dma will issue a completion token when it's done. The `dma_wait` op waits for these completion tokens. So in the code snippet the `dma_wait` might be synchronizing with an earlier token (copy) than intended. This might not be obvious behavior, but it is [documented ](https://xilinx.github.io/mlir-aie/AIEXDialect.html) in the aiex dialect description. Hopefully that explains the behavior you're seeing.

closed 2025-04-07 by victorrjimenezz