Xilinx/mlir-aie #2211: Passing arguments to the JIT'd function

ypapadop-amd, 2025-04-18:
The following would allow passing arguments to the function and optional positional decorator arguments:
```python
import functools

def jit(function=None, is_placed=True, use_cache=True):

    def decorator(fn):

        @functools.wraps(fn)
        def wrapped_function(*args, **kwargs):
            fn(*args, **kwargs) # this would be the compilation / launch phase

        return wrapped_function

    if function:
        return decorator(function)

    return decorator

@jit
def test(v):
    print(v)

@jit(is_placed=False)
def test2():
    print(111)
```

So for the vector add, it'll be something like:
```python
@iron.jit(is_placed=False)
def vector_vector_add(input0, input1, output):
    # infer elements, do error checking etc. using input0, input1, output

def main():
    vector_vector_add(input0, input1, output) # we already do that
```

The drawbacks are
1. one additional wrapping / indirection and
2. the dependency on `functools`.

Thoughts @jgmelber, @mawad-amd?

ypapadop-amd, 2025-04-18:
We can always put the `device` as a decorator argument, I don't know what's the right thing to do here.

mawad-amd, 2025-04-18:
I like having `device` as a decorator argument but I prefer if users don't have to specify it. I am thinking about system-wide variables/functions like (`iron.get_default_device()`) which can be default value for device.

ypapadop-amd, 2025-04-18:
I have a solution that passes everything as local args, device included.

I need to read a little bit more on how to do it as a decorator argument, because if we have a generic enough function that can support multiple devices, we're going to end up doing something like
```python
def f(input, output):
   ...

@iron.jit(device=npu)
def f_npu(*args, **kwargs):
   return (*args, **kwargs);
```
which is not great (esp. for JIT from other frameworks).

PR #2213 (2025-04-25): Passing arguments through iron.jit decorator
This PR allows to pass arguments through the `iron.jit` decorator to the called function.

Closes #2211
closed 2025-04-25 by ypapadop-amd