Xilinx/mlir-aie #2214: Timeline units in Perfetto trace are inconsistent with `std::chrono::high_resolution_clock`

Fangtangtang, 2025-04-22:
I'm currently profiling with trace following the guidance provided in the mlir-aie repository. I generated a trace for the [single-core matrix multiplication example](https://github.com/Xilinx/mlir-aie/tree/fd89c9615bac2e5d71e07bc118d9dc48851ce048/programming_examples/basic/matrix_multiplication/single_core) and visualized it using [ui.perfetto.dev](https://ui.perfetto.dev). However, I’ve noticed that the timeline units shown in Perfetto seem inconsistent with the timing data measured in C++ using `std::chrono::high_resolution_clock`.

I use the following to capture execution time:
```cpp
auto start = std::chrono::high_resolution_clock::now();
// Run kernel
auto stop = std::chrono::high_resolution_clock::now();
float npu_time = std::chrono::duration_cast<std::chrono::microseconds>(stop - start).count();
```
This gives microsecond-level timing, but the Perfetto trace appears to be on the millisecond (ms) scale based on the timeline view.

[image]

For reference, I’m working from commit `fd89c9`, checked out directly via:
```bash
git checkout fd89c9
```

Could you clarify how to read or interpret the timeline scale in Perfetto traces? Is there a recommended method to match the Perfetto trace timestamps with the durations obtained from C++ timing APIs?

Thanks a lot for your help!

jgmelber, 2025-04-22:
Hi! You're correct the timing reported in perfetto is misleading, while it reports ms the unit is cycles. It hope that helps clarify.

Fangtangtang, 2025-04-22:
Thank you for the clarification!

Fangtangtang, 2025-04-23:
Hi again, and thanks again for your help earlier!

I've been trying to reconcile the trace cycles with actual time. Since the trace reports values in cycles, I expected I could estimate the duration using the NPU clock frequency.
For example, in my trace, an event takes about 406 cycles, the duration appears to be around 620 microseconds. I tried estimating the clock frequency from this (which comes out to roughly 655 kHz), but it doesn't seem to match the expected NPU frequency — I thought the NPU clock would be in the GHz range.

Is there a specific or recommended way to convert these cycles into actual time?

Any clarification would be greatly appreciated!

jgmelber, 2025-04-23:
#2017 has information on clock frequencies and tools to discover.

jgmelber, 2025-04-23:
https://github.com/amd/xdna-driver/issues/495

AndraBisca, 2025-06-12:
Closed for inactivity. Please re-open as necessary.

closed 2025-06-12 by AndraBisca