Xilinx/mlir-aie #2218: Matrix multiplication, can't fill the fifo with matrix tiles multiple times

allythe, 2025-04-23:
Hi! I need to make matrix multiplication function. C = A @ B, let's assume that all matrices are 8x8. I want to implement the following logic: create objectFifo with depth 1 for the tile of shape [8,1] for matrices A and B, while for the matrix C create objectFifo with shape [1,1]. In the core function I will do this:

def core_fn(fA, fB, fC, fn):
        total = num_sub_vectors * num_sub_vectors
        for _ in range_(num_sub_vectors):
            a_seg = fA.acquire(1)

            for _ in range_(num_sub_vectors):
                b_seg = fB.acquire(1)
                out   = fC.acquire(1)
                fn(a_seg, b_seg, out)   
                
                fB.release(1)
                fC.release(1)
            fA.release(1)
        
And to acquire several rows of B multuple times, I need to fill the fB with matrix B 8 times. So I do

 rt = Runtime()
    with rt.sequence(image, image, image) as (flt_in, ref_in, mi_out):
        #rt.enable_trace(trace_size, workers=[my_worker])
        rt.start(my_worker)
        rt.fill(of_A.prod(), flt_in)
        
        for i in range(num_sub_vectors):
            rt.fill(of_B.prod(), ref_in)

        rt.drain(of_C.cons(), mi_out, wait=True)

However, when i try to increase the size from 8x8 to 16x16, everything fails and I get this error 
Traceback (most recent call last):
  File "/home/mliraie/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/bin/aiecc.py", line 12, in <module>
    main()
  File "/home/mliraie/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/compiler/aiecc/main.py", line 1351, in main
    run(module_str)
  File "/home/mliraie/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/compiler/aiecc/main.py", line 1325, in run
    asyncio.run(runner.run_flow())
  File "/usr/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/mliraie/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/compiler/aiecc/main.py", line 1131, in run_flow
    npu_insts_module = run_passes_module(
                       ^^^^^^^^^^^^^^^^^^
  File "/home/mliraie/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/compiler/aiecc/main.py", line 347, in run_passes_module
    raise e
  File "/home/mliraie/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/python/aie/compiler/aiecc/main.py", line 344, in run_passes_module
    pm.run(mlir_module.operation)
aie._mlir_libs._site_initialize.<locals>.MLIRError: Failure while executing pass pipeline:
error: "-":154:13: 'aiex.dma_configure_task' op Allocator exhausted available buffer descriptor IDs.
 note: "-":154:13: see current operation: 
  %32 = "aiex.dma_configure_task"(%1) <{channel = 1 : i32, direction = 1 : i32, issue_token = false, repeat_count = 0 : i32}> ({
    "aie.dma_bd"(%arg1) <{burst_length = 0 : i32, dimensions = #aie<bd_dim_layout_array[<size = 1, stride = 0>, <size = 1, stride = 0>, <size = 1, stride = 0>, <size = 256, stride = 1>]>, len = 256 : i32, offset = 0 : i32}> : (memref<256xf32>) -> ()
    "aie.end"() : () -> ()
  }) : (index) -> index
error: "-":159:13: 'aiex.dma_configure_task' op Allocator exhausted available buffer descriptor IDs.
 note: "-":159:13: see current operation: 
  %33 = "aiex.dma_configure_task"(%1) <{channel = 0 : i32, direction = 0 : i32, issue_token = true, repeat_count = 0 : i32}> ({
    "aie.dma_bd"(%arg2) <{burst_length = 0 : i32, dimensions = #aie<bd_dim_layout_array[<size = 1, stride = 0>, <size = 1, stride = 0>, <size = 1, stride = 0>, <size = 256, stride = 1>]>, len = 256 : i32, offset = 0 : i32}> : (memref<256xf32>) -> ()
    "aie.end"() : () -> ()
  }) : (index) -> index
error: "-":160:9: 'aie.dma_bd' op Free called on BD chain with unassigned IDs.
 note: "-":160:9: see current operation: "aie.dma_bd"(%arg2) <{burst_length = 0 : i32, dimensions = #aie<bd_dim_layout_array[<size = 1, stride = 0>, <size = 1, stride = 0>, <size = 1, stride = 0>, <size = 256, stride = 1>]>, len = 256 : i32, offset = 0 : i32}> : (memref<256xf32>) -> ()
error: "-":155:9: 'aie.dma_bd' op Free called on BD chain with unassigned IDs.
 note: "-":155:9: see current operation: "aie.dma_bd"(%arg1) <{burst_length = 0 : i32, dimensions = #aie<bd_dim_layout_array[<size = 1, stride = 0>, <size = 1, stride = 0>, <size = 1, stride = 0>, <size = 256, stride = 1>]>, len = 256 : i32, offset = 0 : i32}> : (memref<256xf32>) -> ()
make: *** [Makefile:46: build/final.xclbin] Error 1

Can you help me to resolve this issue?
If I can not fill the fifo multiply times, how can I access the same row of matrix B num_sub_vectors times?

allythe, 2025-04-24:
Thank you for your quick reply! Here is the code

aie2.py
`
import numpy as np
import argparse
import sys

from aie.iron import Kernel, ObjectFifo, Program, Runtime, Worker
from aie.iron.placers import SequentialPlacer
from aie.iron.device import NPU1Col1, NPU2
from aie.iron.controlflow import range_

# tile_size = in1_size from makefile = width = length of the output matrix
def my_mse(dev, tile_size):
    image_dtype = np.float32
    out_dtype = np.float32
    
    out_size = 1

    num_sub_vectors = tile_size 
    
    # Define tensor types
    image = np.ndarray[(tile_size*tile_size, ), np.dtype[image_dtype]]
    image_tile  = np.ndarray[(tile_size,), np.dtype[image_dtype]]
    out_tile  = np.ndarray[(out_size,), np.dtype[image_dtype]]

    
    # External, binary kernel definition
    my_fn = Kernel(
        "matrix_mult_t",
        "scale.o",
        [image_tile, image_tile, out_tile],
    )

    # Input data movement
    of_A = ObjectFifo(image_tile, default_depth=1, name="float")
    of_B = ObjectFifo(image_tile, default_depth=1, name="ref")
    # Output data movement
    of_C = ObjectFifo(out_tile, default_depth=1, name="out")

    # Task for the core to perform

    def core_fn(fA, fB, fC, fn):
        total = num_sub_vectors * num_sub_vectors
        for _ in range_(num_sub_vectors):
            a_seg = fA.acquire(1)

            for _ in range_(num_sub_vectors):
                b_seg = fB.acquire(1)
                out   = fC.acquire(1)
                fn(a_seg, b_seg, out)   
                
                fB.release(1)
                fC.release(1)
            fA.release(1)
        
    # Create a worker to perform the task
    my_worker = Worker(
        core_fn, [of_A.cons(), of_B.cons(), of_C.prod(), my_fn]
    )

    # Runtime operations to move data to/from the AIE-array
    rt = Runtime()
    with rt.sequence(image, image, image) as (flt_in, ref_in, mi_out):
        #rt.enable_trace(trace_size, workers=[my_worker])
        rt.start(my_worker)
        rt.fill(of_A.prod(), flt_in)
        
       
        for _ in range(num_sub_vectors):
            rt.fill(of_B.prod(), ref_in)
    
        rt.drain(of_C.cons(), mi_out, wait=True)

    # Create the program from the device type and runtime
    my_program = Program(dev, rt)

    # Place components (assign them resources on the device) and generate an MLIR module
    return my_program.resolve_program(SequentialPlacer())

# Check if there are at least 4 arguments (dev, in1_size, in2_size, out_size)
if len(sys.argv) < 5:
    raise ValueError(
        "[ERROR] Need at least 4 arguments (dev, in1_size, in2_size, out_size)"
    )

# Create an argument parser
p = argparse.ArgumentParser()
p.add_argument("-d", "--dev", required=True, dest="device", help="AIE Device")
p.add_argument("-i1s", "--in1_size", required=True, dest="in1_size", help="Input 1 size")
p.add_argument("-i2s", "--in2_size", required=True, dest="in2_size", help="Input 2 size")
p.add_argument("-os", "--out_size", required=True, dest="out_size", help="Output size")
p.add_argument(
    "-t",
    "--trace_size",
    required=False,
    dest="trace_size",
    default=0,
    help="Trace buffer size",
)
opts = p.parse_args(sys.argv[1:])

# Determine the device based on the provided argument
if opts.device == "npu":
    dev = NPU1Col1()
elif opts.device == "npu2":
    dev = NPU2()
else:
    raise ValueError("[ERROR] Device name {} is unknown".format(opts.device))

# Validate the size of the first input
tile_size = int(opts.in1_size)

module = my_mse(dev, tile_size)

# Print the generated MLIR module
print(module)
`

mse.cc
`//===- vector_scaler_mul.cc -------------------------------------*- C++ -*-===//
//
// This file is licensed under the Apache License v2.0 with LLVM Exceptions.
// See https://llvm.org/LICENSE.txt for license information.
// SPDX-License-Identifier: Apache-2.0 WITH LLVM-exception
//
// Copyright (C) 2024, Advanced Micro Devices, Inc.
//
//===----------------------------------------------------------------------===//

#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <type_traits>
#include <aie_api/aie.hpp>

#define vector_size 8// Size of the vector unit, used for reading from input stream
#define read_type uint8 // type of the input stream
#define func_type int // type used for doing the computations
#define write_type float // type of the output stream
#define SIZE 16

extern "C" {

void matrix_mult_t(float* img1, float* img2, float* out) {
  event0();
    float *__restrict p_img1 = img1;
    float *__restrict p_img2 = img2;
    float *__restrict p_out = out;

    aie::vector<float,vector_size> vect1;
    aie::vector<float,vector_size> vect2;
    aie::vector<float,vector_size> mult;
    float partial_sum = 0;

    for(int i = 0; i < SIZE / vector_size; i++) {
        vect1 = aie::load_v<vector_size>(p_img1);
        p_img1 += vector_size;
    
        vect2 = aie::load_v<vector_size>(p_img2);
        p_img2 += vector_size;
    
        mult = aie::mul(vect1, vect2).to_vector<float>(); 
        partial_sum += aie::reduce_add(mult);
        
    }

    p_out[0] = partial_sum;

  
  event1();
}
} // extern "C"
`

allythe, 2025-04-24:
Thank you for your quick reply! Here is the code

aie.py
```python
import numpy as np
import argparse
import sys

from aie.iron import Kernel, ObjectFifo, Program, Runtime, Worker
from aie.iron.placers import SequentialPlacer
from aie.iron.device import NPU1Col1, NPU2
from aie.iron.controlflow import range_

def my_mse(dev, tile_size):
    image_dtype = np.float32
    out_dtype = np.float32

    out_size = 1
    num_sub_vectors = tile_size 

    image = np.ndarray[(tile_size*tile_size,), np.dtype[image_dtype]]
    image_tile  = np.ndarray[(tile_size,), np.dtype[image_dtype]]
    out_tile  = np.ndarray[(out_size,), np.dtype[image_dtype]]

    my_fn = Kernel(
        "matrix_mult_t",
        "scale.o",
        [image_tile, image_tile, out_tile],
    )

    of_A = ObjectFifo(image_tile, default_depth=1, name="float")
    of_B = ObjectFifo(image_tile, default_depth=1, name="ref")
    of_C = ObjectFifo(out_tile, default_depth=1, name="out")

    def core_fn(fA, fB, fC, fn):
        total = num_sub_vectors * num_sub_vectors
        for _ in range_(num_sub_vectors):
            a_seg = fA.acquire(1)
            for _ in range_(num_sub_vectors):
                b_seg = fB.acquire(1)
                out   = fC.acquire(1)
                fn(a_seg, b_seg, out)   
                fB.release(1)
                fC.release(1)
            fA.release(1)

    my_worker = Worker(
        core_fn, [of_A.cons(), of_B.cons(), of_C.prod(), my_fn]
    )

    rt = Runtime()
    with rt.sequence(image, image, image) as (flt_in, ref_in, mi_out):
        rt.start(my_worker)
        rt.fill(of_A.prod(), flt_in)
        for i in range(num_sub_vectors):
            rt.fill(of_B.prod(), ref_in)
        rt.drain(of_C.cons(), mi_out, wait=True)

    my_program = Program(dev, rt)
    return my_program.resolve_program(SequentialPlacer())

if len(sys.argv) < 5:
    raise ValueError(
        "[ERROR] Need at least 4 arguments (dev, in1_size, in2_size, out_size)"
    )

p = argparse.ArgumentParser()
p.add_argument("-d", "--dev", required=True, dest="device", help="AIE Device")
p.add_argument("-i1s", "--in1_size", required=True, dest="in1_size", help="Input 1 size")
p.add_argument("-i2s", "--in2_size", required=True, dest="in2_size", help="Input 2 size")
[... 467 lines cut ...]
/**
 * @brief Main function.
 *
 * Parses command-line arguments, prints the parameters, and launches the kernel execution.
 *
 * @param argc Number of command-line arguments.
 * @param argv Array of command-line argument strings.
 * @return 0 on success, 1 on failure.
 */
int main(int argc, const char *argv[]) {

    constexpr int OUT_N = OUT_SIZE;
    constexpr int IN_N = OUT_N/UPSCALE_FACTOR;

    std::cout << "Going to run the kernel with the following parameters:\n";
    std::cout << "In matrix size: " << IN_N << " by " << IN_N << "\n";
    std::cout << "Upscale factor: " << UPSCALE_FACTOR << "\n";
    
    std::cout << "Out matrix size: " << OUT_N << " by " << OUT_N << "\n";

  args myargs = parse_args(argc, argv);
  int res = setup_and_run_aie(IN_N, OUT_N, myargs);
  return 0;
}
```

jgmelber, 2025-04-28:
Fill and Drain can be called multiple times to push multiple tiles into ObjectFIFOs or pull to write back to main memory. 

For larger matrices it becomes important to manage the fill and drain operations in more detail to properly tile and ensure synchronization: https://github.com/Xilinx/mlir-aie/blob/eb8a0d8e5b4cbdef44514cfa55660b92c1ef4796/programming_examples/basic/matrix_multiplication/single_core/single_core_iron.py#L244-L283

closed 2025-04-30 by jgmelber