Xilinx/mlir-aie #2220: Recommended method to profile performance on Ryzen AI NPU

Fangtangtang, 2025-04-24:
I'm currently developing on **Ryzen AI hardware (NPU1)** and am trying to find a reliable way to measure the performance of a complete program running on the NPU.

According to the [Programming Guide](https://github.com/Xilinx/mlir-aie/tree/main/programming_guide), I have tried the methods in Section 4. I observed that the trace-based timeline view seems useful for analyzing performance on compute tiles. However, I also would like to measure the total runtime of the entire program, not just tile-level kernel execution.
To that end, I used the timer-based method and observed significant variability, even after applying warm-up and averaging multiple runs. I recorded 1000 executions of the same program and observed some short-term stability, but the overall timing profile showed a staircase-like pattern.

[image]

I suspect this may be related to [Performance Level of the NPU](https://ryzenai.docs.amd.com/en/latest/xrt_smi.html#xrt-smi-configure).

Currently, the device appears to be operating in default mode. I attempted to manually switch the performance level using `xrt-smi configure --pmode performance -d <device>`, as mentioned in the [XRT SMI documentation](https://ryzenai.docs.amd.com/en/latest/xrt_smi.html#xrt-smi-configure), but encountered this error:

```
[xrt-smi] ERROR: DRM_IOCTL_AMDXDNA_SET_STATE IOCTL failed (err=95): Operation not supported
```

### Questions:

1. What is the recommended method to measure end-to-end program performance on Ryzen AI (including both host interaction and NPU execution)?
2. Is it expected that the performance mode cannot be changed on Ryzen AI?
3. Are there alternative ways to fix the performance level or otherwise reduce timing variability during benchmarking?

Thanks in advance for your help!

jgmelber, 2025-04-24:
> Questions:
> 
> What is the recommended method to measure end-to-end program performance on Ryzen AI (including both host interaction and NPU execution)?

Using the trace is a great way to get an understanding of the single-core execution and performance of a given kernel. Using timers on the hose with warmup and multiple execution runs is a best practice for performance profiling of the NPU. I will add a note on performance mode below. It may also be advantageous to ensure the CPU does not sleep by running a thread doing some "work" while the NPU is executing. 

> Is it expected that the performance mode cannot be changed on Ryzen AI?

Performance mode should be modifiable on Ryzen AI, however on some platforms this feature is not supported (Phoenix, if I remember correctly). Using turbo mode and ensuring your laptop is plugged in rather than on battery power should help.

> Are there alternative ways to fix the performance level or otherwise reduce timing variability during benchmarking?

There will always be some level of variability in any real system, the DDR memory and interconnect are shared resources that can lead to variability.

jgmelber, 2025-04-24:
@jackl-xilinx do you have anything to add?

chhzh123, 2025-04-24:
Hi @jgmelber, thanks for the prompt answer! We are using the [AMD Ryzen™ 9 7940HS](https://www.amd.com/en/products/processors/laptop/ryzen/7000-series/amd-ryzen-9-7940hs.html) PC for benchmarking. I noticed the former codename is "Phoenix". Is it the platform you mentioned whose performance mode cannot be changed? In that case, are there any ways to get consistent and stable profiling results?

> It may also be advantageous to ensure the CPU does not sleep by running a thread doing some "work" while the NPU is executing.

Can you elaborate a bit more about this? It'd be great if you could point us to a specific example. Thanks!

jgmelber, 2025-04-24:
Yes, there aren't performance modes on that part. You should just use balanced. Try profiling after a fresh restart with nothing else running on the system. It's hard to tell if any other applications are contributing to the jitter. 

> Can you elaborate a bit more about this? It'd be great if you could point us to a specific example. Thanks!

```
#include <iostream>

int main() {
    std::cout << "Starting busy loop. Press Ctrl+C to stop.\n";

    while (1) {
        // Busy wait
        asm(""); // Prevents compiler optimizations
    }

    return 0;
}
```

jackl-xilinx, 2025-04-24:
@jgmelber Nothing to add. Variability in host code is generally expected as it runs within a host execution environment which shares resources with the cpu and operates within a memory cache framework.

jackl-xilinx, 2025-04-25:
Adding @maxzhen to give more insight on performance mode issues on phoenix.

maxzhen, 2025-04-25:
Not sure what version of the driver you're using? I just tried the latest driver on my Pheonix system and I can set performance mode with sudo xrt-smi configure --pmode performance. Anyway, I think 'performance' is the default power mode, so it won't make any difference. I can't say if this level of variation is expected or not, but you will expect some variation for sure. The CPU frequency varies depending on how busy it is since this type of application is not compute bound. The scheduler latency (when thread becomes runnable till it is scheduled on a CPU core) and interrupt latency (when MSI-x is triggered till driver's callback is called) also play a big role here. Basically, the more hops from one thread to another, the more variation you'll see...