Xilinx/mlir-aie #2224: NPU enters corrupt state with large amount of SHIM data transfers

victorrjimenezz, 2025-04-24:
Hello!!

### **Issue**

I'm running into an issue when trying to perform a significant amount of `shim_dma` transfers, where the NPU enters a corrupt state at runtime and throws the following error:

`terminate called after throwing an instance of 'std::runtime_error' `
`what(): qds_device::wait() unexpected command state`

After this, the NPU is no longer functional — any attempt to load a new XCLBIN results in:
`DRM_IOCTL_AMDXDNA_EXEC_CMD IOCTL failed (err=5):`
`Input/output error`

The only solution is to perform a full NPU reset.

This can be easily reproduced in the matrix multiplication whole array example, with large matrices on the NPU. Specifically, when **M = K = N ≥ 2048**.

### **Diagnosis**

After some investigation, it seems the issue is related to the total number of `shim_dma_tasks` being submitted — even if I wait between submissions and ensure that `bd_ids` are available again. Using `npu_dma_memcpy_nd` leads to the same problem. If too many tasks are submitted overall, the NPU enters this broken state.

Here is the relevant code snippet for a matrix-vector multiplication runtime sequence that causes this issue:
`         batch_burst = 1`  
`         batch_size_per_col = batch_burst * m * n_aie_compute_rows`  
`         n_batches = M // (batch_burst * m * n_cores)`  
`         assert M % (batch_burst * m * n_cores) == 0, "This must be divisible."`  

`         in_tasks = []`  
`         out_tasks = []`  
 
`          for batch in range(n_batches): ` 
`              for col in range(n_aie_columns):` 
`                  matrix_in_task = shim_dma_single_bd_task(` 
`                                 alloc=shim_in_matrix[col],` 
`                                 mem=A,` 
`                                 offset=(batch * batch_size_per_col * n_aie_columns * K) + (col * batch_size_per_col * K),` 
`                                 sizes=[1, 1, 1, batch_size_per_col * K],` 
`                                 strides=[m * K, k, K, 1],)` 
`                   dma_start_task(matrix_in_task)` 
`                   in_tasks.append(matrix_in_task)` 
` ` 
`                   vector_in_task = shim_dma_single_bd_task(` 
`                                alloc=shim_in_vector[col],` 
`                                mem=B,` 
`                                offset=0,` 
`                                sizes=[1, 1, 1, (batch_size_per_col // (m * n_aie_compute_rows)) * K],` 
`                                strides=[0, 0, 0, 1],)` 
`                   dma_start_task(vector_in_task)` 
`                   in_tasks.append(vector_in_task)` 
` ` 
`                   vector_out_task = shim_dma_single_bd_task(` 
`                                alloc=shim_out_vector[col],` 
`                                mem=C,` 
`                                offset=(batch * batch_size_per_col * n_aie_columns) + (col * batch_size_per_col),` 
`                                sizes=[1, 1, 1, batch_size_per_col],` 
`                                strides=[0, 0, 0, 1],` 
`                                issue_token=True,)` 
`                   dma_start_task(vector_out_task)` 
`                   out_tasks.append(vector_out_task)` 
`               dma_await_task(*out_tasks)` 
`               dma_free_task(*in_tasks)` 
`               in_tasks = []` 
`               out_tasks = []`

### **Example Scenario**
Given the following configuration:    
- **n_aie_compute_rows = 1**
- **n_aie_columns = 4**
- **n_cores = 4**
- **(M, K) = (4096, 4096)**
- **(m, k) = (64, 64)**

With `batch_burst = 1`, the NPU crashes. However, increasing `batch_burst ≥ 2` reduces the number of total transfers and resolves the issue. This behavior is consistent across other configurations where **M, K ≥ 4096**.

### **Conclusion**

The only viable workaround for now seems to be reducing the number of DMA transfers.

Would love to get your thoughts on:

- Whether this is a known hardware/compiler limitation
- If there's a better way to manage **large** shim data transfers

Thanks a lot for the great work on MLIR-AIE!

jgmelber, 2025-04-25:
https://github.com/Xilinx/mlir-aie/blob/main/programming_guide/section-2/section-2d/DMATasks.md#best-practices-for-data-movement-and-synchronization-with-npu_dma_memcpy_nd

> ```
> terminate called after throwing an instance of 'std::runtime_error' 
> what(): qds_device::wait() unexpected command state
> ```

This message is a NPU hang. The command processor is stalled waiting on a token from a DMA channel. This is likely due to a mismatch in the amount of data sent or received by the operations in the runtime sequence.

victorrjimenezz, 2025-04-25:
Thanks for the clarification! That makes sense. I'm just curious — why would the issue only arise with larger datasets? Specifically, in the case of the Matrix Multiplication example, if the code works fine for smaller matrices, shouldn't it behave the same for larger ones as well?

jgmelber, 2025-04-25:
It might only show up in certain cases. You have to be very careful to match the input and output sizes, along with proper synchronization. Additionally each shimDMA tile has 16 buffer descriptors to use, so you have to be careful about resource allocation. You don't want to reuse a buffer descriptor that may be in use for a transfer. That is where synchronization to guarantee the transfers have completed is essential.

victorrjimenezz, 2025-04-25:
Thanks for the explanation, I understand now. Just to clarify, would this behavior be considered a bug in the [Whole Array Matrix Multiplication Example](https://github.com/Xilinx/mlir-aie/blob/main/programming_examples/basic/matrix_multiplication/whole_array/whole_array.py), or is this something that’s expected due to the resource allocation and synchronization concerns?

jgmelber, 2025-04-25:
I believe you are hitting one of the current limitations of the DMA schedule generated by the parameterized implementation. It is possible to generate alternative schedules that may work better for your use cases.

victorrjimenezz, 2025-04-25:
I see, that explains why generating a few non-parameterized linear transfers works for bigger sizes! Thank you for the insight!

closed 2025-04-25 by victorrjimenezz