Xilinx/mlir-aie #2242: Feature Request: Create a python type binding for "xilinx:AIE::PacketInfoAttr"

joeldushouyu, 2025-04-28:
Hello all,

In the MLIR-python library, the [NpuDmaMemcpyNd](https://github.com/Xilinx/mlir-aie/blob/7bf078158e073ad862e57b8ae63dc287f4faf695/python/dialects/aiex.py#L60)  will translate to [aiex.npu.dma_memcpy_nd](https://xilinx.github.io/mlir-aie/AIEXDialect.html).

Currently, I encountered an issue when trying to called the function to send message in packet_flow.
1. The constructor for [NpuDmaMemcpyND](https://github.com/Xilinx/mlir-aie/blob/7bf078158e073ad862e57b8ae63dc287f4faf695/python/dialects/aiex.py#L60) seems missing parameters for passing packet_id and packet type.

Given the situation, I tried to write my own wrapper function that inheritance NpuDmaMEmcpyNdOp(generated by mlir-tblgen under aie.dialects._aiex_ops_gen

While looking at the code, I found the packet parameter of NpuDmaMemcoyNdOp always result to invalid value(or being ignored) as shown below due to 
```python
_ods_ir.AttrBuilder.contains('PacketInfoAttr'))
``` 
not being defined.

```python
@_ods_cext.register_operation(_Dialect)
class NpuDmaMemcpyNdOp(_ods_ir.OpView):
  OPERATION_NAME = "aiex.npu.dma_memcpy_nd"

  _ODS_OPERAND_SEGMENTS = [1,-1,-1,-1,]

  _ODS_REGIONS = (0, True)

  def __init__(self, memref, offsets, sizes, strides, static_offsets, static_sizes, static_strides, metadata, id, *, packet=None, issue_token=None, d0_zero_before=None, d1_zero_before=None, d2_zero_before=None, d0_zero_after=None, d1_zero_after=None, d2_zero_after=None, burst_length=None, loc=None, ip=None):
    operands = []
    results = []
    attributes = {}
    regions = None
    operands.append(memref)
    operands.append(_get_op_results_or_values(offsets))
    operands.append(_get_op_results_or_values(sizes))
    operands.append(_get_op_results_or_values(strides))
    _ods_context = _ods_get_default_loc_context(loc)
    attributes["static_offsets"] = (static_offsets if (
    isinstance(static_offsets, _ods_ir.Attribute) or
    not _ods_ir.AttrBuilder.contains('DenseI64ArrayAttr')) else
      _ods_ir.AttrBuilder.get('DenseI64ArrayAttr')(static_offsets, context=_ods_context))
    attributes["static_sizes"] = (static_sizes if (
    isinstance(static_sizes, _ods_ir.Attribute) or
    not _ods_ir.AttrBuilder.contains('DenseI64ArrayAttr')) else
      _ods_ir.AttrBuilder.get('DenseI64ArrayAttr')(static_sizes, context=_ods_context))
    attributes["static_strides"] = (static_strides if (
    isinstance(static_strides, _ods_ir.Attribute) or
    not _ods_ir.AttrBuilder.contains('DenseI64ArrayAttr')) else
      _ods_ir.AttrBuilder.get('DenseI64ArrayAttr')(static_strides, context=_ods_context))
    if packet is not None: attributes["packet"] = (packet if (
        isinstance(packet, _ods_ir.Attribute) or
        not _ods_ir.AttrBuilder.contains('PacketInfoAttr')) else
          _ods_ir.AttrBuilder.get('PacketInfoAttr')(packet, context=_ods_context))
    attributes["metadata"] = (metadata if (
    isinstance(metadata, _ods_ir.Attribute) or
    not _ods_ir.AttrBuilder.contains('FlatSymbolRefAttr')) else
      _ods_ir.AttrBuilder.get('FlatSymbolRefAttr')(metadata, context=_ods_context))
```

Currently, I have resolved the issue by calling mlir.Attribute.parse function to generate the corresponding "PacketInfoAttr" based on the definition[https://github.com/Xilinx/mlir-aie/blob/7bf078158e073ad862e57b8ae63dc287f4faf695/include/aie/Dialect/AIE/IR/AIEAttrs.td#L193]. I have attached my solution below. Also, the **dma_bd() python binding function** also suffers a similar issue. 

Obviously, the solution is a temporary and I think it could better if there is build in type-binding for it? I am happy to create a PR for if if necessary.

```python

from aie.dialects.transform.structured import MixedValues, _dispatch_mixed_values
from aie.dialects._aiex_ops_gen import NpuDmaMemcpyNdOp, npu_dma_memcpy_nd
from aie._mlir_libs._mlir.ir import DenseF64ArrayAttr

from aie.dialects._ods_common import get_default_loc_context as _ods_get_default_loc_context
from aie.dialects._ods_common  import _cext as _ods_cext
_ods_ir = _ods_cext.ir

from aie._mlir_libs._mlir.ir import Attribute
from aie.dialects._aiex_ops_gen import _Dialect

def generate_packet_attribute(packet_id:int, packet_type:int):
    return Attribute.parse(f"#aie.packet_info<pkt_type = {packet_type}, pkt_id = {packet_id}>")
class NpuDmaMemcpyNd(NpuDmaMemcpyNdOp):

    def __init__(
        self,
        metadata: str ,
        bd_id,
        mem,
        tap:   None = None,
        offsets: MixedValues | None = None,
        sizes: MixedValues | None = None,
        strides: MixedValues | None = None,
        issue_token: bool | None = None,
        burst_length: int = 0,
        packet_id = None,
        packet_type = None 
    ):
        if tap and not (offsets is None and sizes is None and strides is None):
            raise ValueError(
                "NpuDmaMemcpyNd can take either a TileAccessPattern OR (sizes and/or strides and/or offsets), but not both."
            )
        if tap:
            sizes = tap.sizes.copy()
            strides = tap.strides.copy()
            # For some reason, the type checking of offsets does not mesh well with offset being a property
            # so here we make sure it is evaluated and properly is seen as an integer.
            offsets = [0] * 3 + [int(tap.offset)]
        else:
            if offsets is None:
                offsets = [0] * 4
            if sizes is None:
                sizes = [0] * 4
            if strides is None:
                strides = [0] * 3 + [1]
        dynamic_offsets, _packed_offsets, static_offsets = _dispatch_mixed_values(
            offsets
        )
        dynamic_sizes, _packed_sizes, static_sizes = _dispatch_mixed_values(sizes)
        dynamic_strides, _packed_strides, static_strides = _dispatch_mixed_values(
            strides
        )
        if packet_id is not None and packet_type is not None:
            packet =generate_packet_attribute(packet_id=packet_id, packet_type=packet_type)
        else:
            packet = None
        super().__init__(
            mem,
            dynamic_offsets,
            dynamic_sizes,
            dynamic_strides,
            static_offsets,
            static_sizes,
            static_strides,
            metadata,
            bd_id,
            issue_token=issue_token,
            burst_length=burst_length,
            packet = packet

        )
 

```

jgmelber, 2025-04-28:
#2235 tracks support for Packet switching in the IRON Python API

jgmelber, 2025-04-28:
> I am happy to create a PR for if if necessary

That would be great!

jgmelber, 2025-04-28:
@AndraBisca can you help answer questions related to the Python bindings and IRON library?

joeldushouyu, 2025-04-29:
Hello @AndraBisca , 

Please correct me if I am wrong.

From inspecting the code base, it seems to me for that types defined with cppNamespace Attribute such as [AIEDevice](https://github.com/Xilinx/mlir-aie/blob/main/include/aie/Dialect/AIE/IR/AIEAttrs.td#L100) and [WireBundle](https://github.com/Xilinx/mlir-aie/blob/main/include/aie/Dialect/AIE/IR/AIEAttrs.td#L54C5-L54C15) will generate a corresponding python type binding during compile stage through pybinding.

But for types such as [BDPadLayoutArrayAttr](https://github.com/Xilinx/mlir-aie/blob/main/include/aie/Dialect/AIE/IR/AIEAttrs.td#L186) or [AIEI64Attr](https://github.com/Xilinx/mlir-aie/blob/main/include/aie/Dialect/AIE/IR/AIEAttrs.td#L35), there exist a manual type declaration for [BDpadLayoutArrayAttr](https://github.com/Xilinx/mlir-aie/blob/main/python/dialects/aie.py#L136) and [aIEI64Atttr](https://github.com/Xilinx/mlir-aie/blob/main/python/dialects/aie.py#L167) in aie.py.

Thus, given that [PacketInfoAttr](https://github.com/Xilinx/mlir-aie/blob/main/include/aie/Dialect/AIE/IR/AIEAttrs.td#L193) is also a declaration without defining the cppNamespace attribute, I should just add a similar type-declaration function in [aie.py](https://github.com/Xilinx/mlir-aie/blob/main/python/dialects/aie.py#L118) ?

Moreover, I wonder is there a script (similar to the install script) that I can build the codebase locally to verify before submit the PR?

Thanks a lot!

jgmelber, 2025-04-29:
Instructions for building locally: https://github.com/Xilinx/mlir-aie/blob/main/docs/Building.md

AndraBisca, 2025-04-30:
Hello @joeldushouyu,

Thank you for reaching out and for your additions!

Regarding the type definition, your understanding is very good. Please add it to aie.py.

Let us know when your PR is up and we will review! Thank you!

joeldushouyu, 2025-04-30:
> Hello [@joeldushouyu](https://github.com/joeldushouyu),
> 
> Thank you for reaching out and for your additions!
> 
> Regarding the type definition, your understanding is very good. Please add it to aie.py.
> 
> Let us know when your PR is up and we will review! Thank you!

Thanks,
I will try to come up with a PR sometime this weekend for it.

closed 2025-05-28 by AndraBisca