Xilinx/mlir-aie #2247: Use Peano to build mlir-aie from source

jgmelber, 2025-04-29:
Remove dependency on Vitis to build mlir-aie toolchain

jgmelber, 2025-04-29:
CC @stephenneuendorffer @denolf

jgmelber, 2025-05-07:
Testing this today, it seems like it is fine to build mlir-aie without Vitis. I was unable to get `ninja check-reference-designs` to run as it reported all Unsupported.

fifield, 2025-05-07:
> Testing this today, it seems like it is fine to build mlir-aie without Vitis. I was unable to get `ninja check-reference-designs` to run as it reported all Unsupported.

The culprit might be the `lit.local.cfg` files, which gate the tests based on which vitis components were detected:
```python
if 'AIE2' not in config.vitis_components:
    config.unsupported = True
``` 
Since none were detected, the tests are marked as Unsupported. This needs to get revisited for Vitis free builds.

jgmelber, 2025-05-07:
Right. This is in the AIE Tools cmake:

https://github.com/Xilinx/cmakeModules/blob/faf9872aa93a7540becf788d1c06c9aa67972d3c/FindAIETools.cmake#L152-L174

jgmelber, 2025-05-07:
WIP https://github.com/Xilinx/mlir-aie/commit/38da27762527c186036ada422af48dd77ca7487a but a potential approach...many npu-xrt tests would need an update:

https://github.com/Xilinx/mlir-aie/blob/38da27762527c186036ada422af48dd77ca7487a/programming_examples/lit.cfg.py#L147-L158

jgmelber, 2025-05-07:
#2288

hunhoffe, 2026-10-26:
The core ask here — building mlir-aie without a Vitis dependency — has been working since the original investigation in this thread (confirmed by @jgmelber in 2025). What's left is exactly the long tail of individual tests/examples that still require Chess/Vitis, which  is now tracked with file-by-file granularity in #3479 ("Tests that still require Chess after #3451, and what each one is waiting on"). Closing this one as superseded — #3479 is the actively-maintained continuation of this effort.

closed 2026-10-26 by hunhoffe