Xilinx/mlir-aie #2258: Inquiring the clock frequency of NPU Phoenix

antonio-fc, 2025-05-02:
Hi team,

May I ask the clock frequency of the NPU Phoenix in the AMD Ryzen 87000?

Thanks!

jgmelber, 2025-05-02:
#2017

closed 2025-05-02 by jgmelber

antonio-fc, 2025-05-05:
But do these numbers also apply to the Phoenix? From what I understand issue #2017 is for the Strix and Hawk models, which belong to the 300 series, while the Phoenix is an older model. Am I wrong?

jgmelber, 2025-05-05:
That issue has information about tools and sources to find the clock frequency for all platforms. 

Phoenix is Ryzen AI 7040
Hawk is Ryzen AI 8040, 8000G and Ryzen 200
Strix is Ryzen AI 300
Krackan is Ryzen AI 350 and 340

You can find more information on the AMD product pages: https://www.amd.com/en/products/processors/consumer/ryzen-ai.html

antonio-fc, 2025-05-08:
I'm actually working with this model: https://www.amd.com/es/products/processors/desktops/ryzen/8000-series/amd-ryzen-7-8700g.html

Also, I have found that this model is stuck with using the Performance power mode. Is this correct?

Which set of clock speeds linked in #2017 (there is one linked for Hawkpoint and one for Strix) would match this model and power mode?

melberj, 2025-05-08:
That device is a Hawk Point

antonio-fc, 2025-05-12:
But in the description (https://www.amd.com/es/products/processors/desktops/ryzen/8000-series/amd-ryzen-7-8700g.html), it says it is a Phoenix.
Or is it that HawkPoint and Phoenix are the same in this aspect?
Also, is the largest number from the list below for the 'performance' or 'turbo' powermode?