Xilinx/mlir-aie #2282: Tracing tool units

antonio-fc, 2025-05-07:
Hi team,

I am using the tracing function to get the execution time of the tile kernels. In Perfetto, it shows the slots in ms/us. However, this issue [https://github.com/Xilinx/mlir-aie/issues/2214], says that the units are actually execution cycles, which makes sense given the numbers I am getting. One thing I find confusing though is, how is it that there are decimal (with decimal point) values when the units are supposed to be cycles? I suspect that the original units (cycles) are interpreted as microseconds, and then converted to milliseconds by Perfetto, but I want to make sure. See the image below as reference:

[image]

jackl-xilinx, 2025-05-07:
@antonio-fc This is because perfetto allows higher precision in the measurement tools and does not snap to whole values in us. If you were to more carefully align your measurement tool to the exact edge of the events, then you would get whole value measurements. For the most part, if you're close in your measurement alignment to the edge of the events, then you can safely round to the nearest whole number.

antonio-fc, 2025-05-07:
So to get the cycles for "Wall duration" at the bottom I would convert the value from miliseconds to microseconds, which in reality are cycles, correct?

jackl-xilinx, 2025-05-07:
@antonio-fc Yes, that's right. So the image if rounded to the edges of events would be 10,943 cycles.

closed 2025-05-08 by jackl-xilinx