Xilinx/mlir-aie #2299: chess-llvm-link issue

derekharker, 2025-05-09:
MLIR compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:00 0/1 1 Worker/tools/Xilinx/Vitis/2024.2/aietools/tps/lnx64/target/bin/LNa64bin/chess-llvm-link: No such file or directory

This is happening when I try running make on tutorial 1 of mlir-tutorials. The file exists in that directory

L-roro, 2025-05-15:
Hi Derek. Thank you for bringing this to our attention.

Assuming that you installed IRON using the wheels as described in the main README, the issue you are likely facing is that the wheels do not currently support the AIE1 architecture that is being used in these tutorials. While we work on updating these tutorials with a more solid fix, you could try to add this as a workaround on your aie.mlir file:
```
module @tutorial_1 {
  aie.device(npu2) {
    // Rest of the mlir code
  }
}
```

derekharker, 2025-05-21:
I've tried that on the tutorial-1 aie.mlir code and it is still giving me the same chess-llvm-link error

L-roro, 2025-05-21:
Hi Derek. In that case, I am unfortunately not able to replicate your exact error with the setups I have tried. I would need more information about how you built mlir-aie and the machine you are using to be able to help you.

Additionally, given that you have already checked that chess-llvm-link does exist in the correct directory, it would be helpful to know what exact file you are missing. You should normally have an error message indicating you the command that failed, similar to:
`Error encountered while running: {your path}/Vitis/2024.2/aietools/tps/lnx64/target/bin/LNa64bin/chess-llvm-link {your path}/mlir_tutorials/tutorial-1/aie.mlir.prj/input.llchesshack.ll {your path}/ironenv/lib/python3.10/site-packages/mlir_aie/aie_runtime_lib/AIE2/chess_intrinsic_wrapper.ll -S -o {your path}/mlir_tutorials/tutorial-1/aie.mlir.prj/input.llchesslinked.ll`

You can then run that command again using strace to find out what file is truly giving you this issue. I asked you to change the mlir code above to npu2 in order to change the path `chess_intrinsic_wrapper.ll` is being searched on.

Also note that, even if you are able to get past the first mlir-tutorial, you may encounter other roadblocks in some of the exercises, since the team is currently discussing how we want to address some issues with these tutorials. If you are just starting out with mlir-aie, you should look into the programming_guide instead.

derekharker, 2025-05-22:
I am just getting started with mlir-aie, but I am using a VCK190 and the programming guide is geared towards NPU dev.

My setup was sourcing my Vitis and xrt tools, and then activating the ironenv, and then env_setup.sh 
I've tried with utils/build-mlir-aie-from-wheels.sh but it doesn't give me much luck.

'strace make
execve("/usr/bin/make", ["make"], 0x7fffdd967c70 /* 81 vars */) = 0
brk(NULL)                               = 0x645f5d453000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7c5f8a182000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/lib/glibc-hwcaps/x86-64-v3/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/lib/glibc-hwcaps/x86-64-v3/", 0x7ffd784c8250, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/lib/glibc-hwcaps/x86-64-v2/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/lib/glibc-hwcaps/x86-64-v2/", 0x7ffd784c8250, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/lib/", {st_mode=S_IFDIR|0775, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/opt/xilinx/xrt/lib/glibc-hwcaps/x86-64-v3/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/xilinx/xrt/lib/glibc-hwcaps/x86-64-v3/", 0x7ffd784c8250, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/xilinx/xrt/lib/glibc-hwcaps/x86-64-v2/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/xilinx/xrt/lib/glibc-hwcaps/x86-64-v2/", 0x7ffd784c8250, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/xilinx/xrt/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/xilinx/xrt/lib/", {st_mode=S_IFDIR|0775, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=114067, ...}) = 0
mmap(NULL, 114067, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7c5f8a166000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\220\243\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
fstat(3, {st_mode=S_IFREG|0755, st_size=2125328, ...}) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
mmap(NULL, 2170256, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7c5f89e00000
mmap(0x7c5f89e28000, 1605632, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x28000) = 0x7c5f89e28000
mmap(0x7c5f89fb0000, 323584, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1b0000) = 0x7c5f89fb0000
mmap(0x7c5f89fff000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1fe000) = 0x7c5f89fff000
mmap(0x7c5f8a005000, 52624, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7c5f8a005000
close(3)                                = 0
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7c5f8a163000
arch_prctl(ARCH_SET_FS, 0x7c5f8a163740) = 0
set_tid_address(0x7c5f8a163a10)         = 1288673
set_robust_list(0x7c5f8a163a20, 24)     = 0
rseq(0x7c5f8a164060, 0x20, 0, 0x53053053) = 0
mprotect(0x7c5f89fff000, 16384, PROT_READ) = 0
mprotect(0x645f2b349000, 8192, PROT_READ) = 0
mprotect(0x7c5f8a1c0000, 8192, PROT_READ) = 0
prlimit64(0, RLIMIT_STACK, NULL, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
munmap(0x7c5f8a166000, 114067)          = 0
getrandom("\x98\x52\x54\xa0\x91\xc5\x53\x7d", 8, GRND_NONBLOCK) = 8
brk(NULL)                               = 0x645f5d453000
brk(0x645f5d474000)                     = 0x645f5d474000
openat(AT_FDCWD, "/usr/lib/locale/locale-archive", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=5719296, ...}) = 0
mmap(NULL, 5719296, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7c5f89800000
close(3)                                = 0
rt_sigaction(SIGHUP, {sa_handler=0x645f2b3202c0, sa_mask=[HUP], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7c5f89e45330}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
rt_sigaction(SIGQUIT, {sa_handler=0x645f2b3202c0, sa_mask=[QUIT], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7c5f89e45330}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
rt_sigaction(SIGINT, {sa_handler=0x645f2b3202c0, sa_mask=[INT], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7c5f89e45330}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
rt_sigaction(SIGTERM, {sa_handler=0x645f2b3202c0, sa_mask=[TERM], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7c5f89e45330}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
rt_sigaction(SIGXCPU, {sa_handler=0x645f2b3202c0, sa_mask=[XCPU], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7c5f89e45330}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
[... cut ...]
openat(AT_FDCWD, "/usr/share/locale-langpack/en/LC_MESSAGES/make.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
write(2, "make: *** [Makefile:15: core_1_4"..., 46make: *** [Makefile:15: core_1_4.elf] Error 1
) = 46
rt_sigprocmask(SIG_BLOCK, [HUP INT QUIT TERM XCPU], NULL, 8) = 0
rt_sigprocmask(SIG_UNBLOCK, [HUP INT QUIT TERM XCPU], NULL, 8) = 0
chdir("/home/dwharker/mlir-aie/mlir_tutorials/tutorial-1") = 0
close(1)                                = 0
exit_group(2)                           = ?
+++ exited with 2 +++'

L-roro, 2025-05-26:
Hi again Derek. It looks like you have called strace on the make command instead of the offending one I described above. The command that is failing is being executed on a subprocess and calling a naked strace on the make command does not output the information needed to identify the missing file or directory. You can either call strace on the failing command or alternatively call strace with the -f flag to output this information. Note that the output will be very long with the -f flag!

derekharker, 2025-05-27:
Error encountered while running: /tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/chess-llvm-link /home/dwharker/mlir-aie/mlir_tutorials/tutorial-1/aie.mlir.prj/input.llchesshack.ll /home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/aie_runtime_lib/AIE/chess_intrinsic_wrapper.ll -S -o /home/dwharker/mlir-aie/mlir_tutorials/tutorial-1/aie.mlir.prj/input.llchesslinked.ll

make: *** [Makefile:15: core_1_4.elf] Error 1
(ironenv) dwharker@goeders7:~/mlir-aie/mlir_tutorials/tutorial-1$ strace /tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/chess-llvm-link /home/dwharker/mlir-aie/mlir_tutorials/tutorial-1/aie.mlir.prj/input.llchesshack.ll /home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/aie_runtime_lib/AIE/chess_intrinsic_wrapper.ll -S -o /home/dwharker/mlir-aie/mlir_tutorials/tutorial-1/aie.mlir.prj/input.llchesslinked.ll
execve("/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/chess-llvm-link", ["/tools/Xilinx/Vitis/2023.2/aieto"..., "/home/dwharker/mlir-aie/mlir_tut"..., "/home/dwharker/mlir-aie/ironenv/"..., "-S", "-o", "/home/dwharker/mlir-aie/mlir_tut"...], 0x7fffb48f60d8 /* 66 vars */) = 0
brk(NULL)                               = 0x1c8b8000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x745eae6c5000
readlinkat(AT_FDCWD, "/proc/self/exe", "/tools/Xilinx/Vitis/2023.2/aieto"..., 4096) = 81
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/glibc-hwcaps/x86-64-v3/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/glibc-hwcaps/x86-64-v3/", 0x7ffe3dc28d30, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/glibc-hwcaps/x86-64-v2/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/glibc-hwcaps/x86-64-v2/", 0x7ffe3dc28d30, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/", {st_mode=S_IFDIR|0755, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/../lib/glibc-hwcaps/x86-64-v3/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/../lib/glibc-hwcaps/x86-64-v3/", 0x7ffe3dc28d30, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/../lib/glibc-hwcaps/x86-64-v2/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/../lib/glibc-hwcaps/x86-64-v2/", 0x7ffe3dc28d30, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/../lib/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/../lib/", 0x7ffe3dc28d30, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/lib/glibc-hwcaps/x86-64-v3/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/lib/glibc-hwcaps/x86-64-v3/", 0x7ffe3dc28d30, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/lib/glibc-hwcaps/x86-64-v2/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/lib/glibc-hwcaps/x86-64-v2/", 0x7ffe3dc28d30, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/lib/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/lib/", {st_mode=S_IFDIR|0775, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/opt/xilinx/xrt/lib/glibc-hwcaps/x86-64-v3/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/xilinx/xrt/lib/glibc-hwcaps/x86-64-v3/", 0x7ffe3dc28d30, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/xilinx/xrt/lib/glibc-hwcaps/x86-64-v2/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/xilinx/xrt/lib/glibc-hwcaps/x86-64-v2/", 0x7ffe3dc28d30, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/xilinx/xrt/lib/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/xilinx/xrt/lib/", {st_mode=S_IFDIR|0775, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
[... cut ...]
openat(AT_FDCWD, "/home/dwharker/mlir-aie/ironenv/lib/python3.12/site-packages/mlir_aie/aie_runtime_lib/AIE/chess_intrinsic_wrapper.ll", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
write(2, "/tools/Xilinx/Vitis/2023.2/aieto"..., 83/tools/Xilinx/Vitis/2023.2/aietools/tps/lnx64/target/bin/LNa64bin/chess-llvm-link: ) = 83
write(2, "No such file or directory", 25No such file or directory) = 25
write(2, "\n", 1
)                       = 1
exit_group(1)                           = ?
+++ exited with 1 +++

When I run the command that failed with the strace command

derekharker, 2025-06-05:
Building MLIR from source has helped this work. Thank you

closed 2025-06-05 by derekharker