Xilinx/mlir-aie #2303: Move data to the mem tile only once

allythe, 2025-05-12:
Good evening!
I am working with matrix multiplication example, which fills fifoB multiple times with the B matrix. Data is moved like this: L3 mem -> shim tile -> mem tile -> compute tile. And if I fill the fifo with B multiple times, it is moved from L3 mem multiple times. Is there a way to move it to the mem tile and skip the L3 mem -> shim tile data movement from all the fillings of the fifo? So that the data will not be moved from L3 mem multiple times, instead of it it will be moved inside the aie.

Best wishes,
Alisa

allythe, 2025-05-14:
Thank you for your answer!
Also, I wanted to ask what is happening in [this line of code](https://github.com/Xilinx/mlir-aie/blob/933364b09ee89db0f5998755106fffe7d838f8ac/programming_examples/basic/matrix_multiplication/single_core/single_core_placed.py#L383). Am I right that here I will send all the B matrix to the fifo inB? Will it be moved from the CPU multiple times in this case or just once?

AndraBisca, 2025-05-21:
`b_task = shim_dma_single_bd_task(inB, B, tap=b_tap)`
In this line of code, the B matrix will indeed be moved to the fifo inB via a shim DMA. inB is instantiated as 
`inB = object_fifo("inB", shim_tile, mem_tile, 2, b_ty)` 
where `b_ty` represents the object size and datatype. `B_sz` is the size of the B host buffer which is bigger than `b_ty` and as such multiple data transfers will be required to store the data in the memtile.

closed 2025-05-28 by allythe