Xilinx/mlir-aie #2316: Outdated mlir_tutorials

L-roro, 2025-05-15:
### The problem
The current mlir_tutorials are not supported by the wheels that are used during installation. Calling aiecc.py will result in a call to site-packages/mlir_aie/aie_runtime_lib/AIE/chess_intrinsic_wrapper.ll. The wrapper is instead in the AIE2 or AIE2p folders.

### The original issue
This is currently resulting in a cryptic "No such file or directory" error on chess-llvm-link: #2299 

### Possible fixes
- aiecc.py should return an error when such a problem with an invalid compiler and architecture combination occurs
- The tutorial docs could be updated to make the requirements clear
- Since peano is replacing chess, it might be time to update these tutorials to use peano instead

derekharker, 2025-05-28:
Any time I try to use aiecc.py it reverts back to chess and I don't know how to compile any of my mlir files into usable files for the VCK190 board.

AndraBisca, 2025-05-30:
Building mlir-aie in a way that is compatible with VCK190 boards is not supported through the wheels. In the past, we used either [this](https://github.com/Xilinx/mlir-aie/blob/main/utils/build-mlir-aie.sh) script or [this](https://github.com/Xilinx/mlir-aie/blob/main/utils/build-mlir-aie-cross.sh) one to build from source. You may be able to get LLVM from a wheel, but I haven't tried it in combination with building mlir-aie from source. A sysroot is also required for VCK190 boards. We used to have documentation for this process, however I can't seem to find it easily. I am looking into it.

derekharker, 2025-06-05:
This has worked for me! Thank you.