Xilinx/mlir-aie #2320: 4K Square GEMM Whole Array Timeout

iavssw, 2025-05-18:
On the most recent commit: ea1e9713b8d75752f2e527a6391ed70b609e4e49 (May 17th) for the whole array matrix multiply example times out for the following configuration

# M, K, N: final GEMM dimensions
M?=4096
K?=4096
N?=4096

# m, k, n: single AIE core GEMM dimensions
m?=64
k?=64
n?=64

n_aie_cols?=4
b_col_maj?=0

kernels=mm_${m}x${k}x${n}
aieargs+=-m $m -k $k -n $n --n-aie-cols ${n_aie_cols} --b-col-maj ${b_col_maj}
runargs+=--b_col_maj ${b_col_maj}
target_suffix=${M}x${K}x${N}_${m}x${k}x${n}_${n_aie_cols}c
use_placed?=0
use_iron?=0

dtype_in?=bf16
dtype_out?=bf16
trace_size?=65536

export XRT_HACK_UNSECURE_LOADING_XCLBIN=1 && \
 ./whole_array.exe -x build/final_4096x4096x4096_64x64x64_4c.xclbin -i build/insts_4096x4096x4096_64x64x64_4c.txt -k MLIR_AIE -M 4096 -K 4096 -N 4096 --b_col_maj 0
Kernel did not complete. Returned status: 6
make: *** [/home/greg/Desktop/mlir-aie/programming_examples/basic/matrix_multiplication/whole_array/../makefile-common:146: run] Error 1
(ironenv) greg@aieHP:~/Desktop/mlir-aie/programming_examples/basic/matrix_multiplication/whole_array$

andrej, 2025-05-19:
Thank you for reporting this. I will look into this and get back to you.

andrej, 2025-05-20:
Hi again! I was not able to reproduce the timeout; the 4096x4096x4096 design, with 64x64x64 tile size and bf16 input and output data types passed verification on my end. Here is what I ran, so you can try to copy my steps:

```
M=4096 K=4096 N=4096 m=64 k=64 n=64 dtype_in=bf16 dtype_out=bf16 make
M=4096 K=4096 N=4096 m=64 k=64 n=64 dtype_in=bf16 dtype_out=bf16 make run
```

Output of the second command:
```
(ironenv) andre@andre-vivobook:~/mlir-aie/programming_examples/basic/matrix_multiplication/whole_array$ M=4096 K=4096 N=4096 m=64 k=64 n=64 dtype_in=bf16 dtype_out=bf16 make run
export XRT_HACK_UNSECURE_LOADING_XCLBIN=1 && \
 ./whole_array.exe -x build/final_4096x4096x4096_64x64x64_4c.xclbin -i build/insts_4096x4096x4096_64x64x64_4c.txt -k MLIR_AIE -M 4096 -K 4096 -N 4096 --b_col_maj 0

Avg NPU matmul time: 75526us.
Avg NPU gflops: 1819.76

Min NPU matmul time: 75526us.
Max NPU gflops: 1819.76

Max NPU matmul time: 75526us.
Min NPU gflops: 1819.76

PASS!
```

If you are still seeing the issue, could you please try the following?

1. Ensure that the compiler commit and repository commit match, as such (in my case `c105c0b..` in both the git log output and aie-opt output)
```
(ironenv) andre@andre-vivobook:~/mlir-aie/programming_examples/basic/matrix_multiplication/whole_array$ git log --oneline -n1
c105c0b6 (HEAD -> main, origin/main, origin/HEAD) Capturing bootgen output and print it iff verbose (#2321)
(ironenv) andre@andre-vivobook:~/mlir-aie/programming_examples/basic/matrix_multiplication/whole_array$ aie-opt --version
LLVM (http://llvm.org/):
  LLVM version 21.0.0git
  Optimized build with assertions.
aie-opt c105c0b657a9599487e1b0bad8f5cdb5377d5acd
```
If they do not match, check out the repository at the same commit as the compiler commit, so the examples match up: `git checkout <aie-opt commit version>`

2. Ensure you have a clean build:
```
make clean
```

3. Ensure you run the host-side code with the correct, matching arguments. It's easiest to run `make run` with the same environment variables as you used for the build.