Xilinx/mlir-aie #2326: programming_guide compatibility with XCVC1902

derekharker, 2025-05-20:
I'm working with a VCK190 board and although the documentation says Versal is supported when I try and change where it has "NPU1Col1" for example in the aie2.py files, it fails. This is an example of an error I get:

    my_program = Program(XCVC1902(), rt)
                         ^^^^^^^^^^
TypeError: Can't instantiate abstract class XCVC1902 without an implementation for abstract methods 'get_compute_tiles', 'get_mem_tiles', 'get_num_dest_shim_mux_connections', 'get_num_dest_switchbox_connections', 'get_num_source_shim_mux_connections', 'get_num_source_switchbox_connections', 'get_shim_tiles'
make: *** [Makefile:15: build/aie.mlir] Error 1

derekharker, 2025-05-21:
Okay that makes more sense! For the placed variant of IRON, is there a specific list of utils that I need to run to set that up? Or where would I go to find that?
Thanks for sending that example file!

derekharker, 2025-05-21:
~/mlir-aie/programming_examples/basic/passthrough_kernel$ make
mkdir -p build
cd build && aiecc.py --aie-generate-xclbin --aie-generate-npu-insts --no-compile-host \
        --no-xchesscc --no-xbridge \
        --xclbin-name=final_in1_size.xclbin --npu-insts-name=insts_in1_size.bin ../build/aie2_lineBased_8b_in1_size.mlir
python3: /__w/mlir-aie/mlir-aie/lib/Targets/AIETargetCDODirect.cpp:145: llvm::LogicalResult translateToCDODirect(mlir::ModuleOp, llvm::StringRef, byte_ordering, bool, bool, bool, bool, bool): Assertion `targetModel.hasProperty(AIETargetModel::IsNPU) && "Only NPU currently supported"' failed.
Aborted (core dumped)
make: *** [Makefile:56: build/final_in1_size.xclbin] Error 134

I'm also getting this error on that file after I changed it. Not sure it's supported yet?

melberj, 2025-05-26:
AIE1 devices use a different [target backend](https://github.com/Xilinx/mlir-aie/blob/main/lib/Targets/AIETargetXAIEV2.cpp).

They also require a platform design and runtime that supports MLIR-AIE an example of this for VCK5000 can be found [here](https://github.com/Xilinx/ROCm-air-platforms).

AndraBisca, 2025-05-30:
Yes, the VCK190 is an AIE1 device. However, as our main focus for the past years has been on developing IRON to target NPU devices, support for AIE1 across the tools and documentation is outdated. In the past, we used to run designs on VCK190 using xchesscc.

AndraBisca, 2025-05-30:
If you want to try and adapt designs written for the NPU for VCK190, you should remove any mention to the runtime sequence as well as compilation flags like `--aie-generate-xclbin` and `--aie-generate-npu-insts` as these are specific to the NPU architecture.