Xilinx/mlir-aie #2333: using all 5 columns

antonio-fc, 2025-05-22:
Hi,

When I started working with MLIR-AIE, the use of "AIEDevice.npu1" was not available (the one with all 5 columns). However, in my search for documentation to justify only using 4 columns in my designs (AIEDevice.npu1_4col), I found examples of the former in the issues of the repository. Is this supported now? 

I tried using it but I got the error:

"
terminate called after throwing an instance of 'xrt_core::system_error'
what():  DRM_IOCTL_AMDXDNA_CREATE_CTX IOCTL failed (err=28): No space left on device
"

Does this mean that I am doing something wrong? Or is this a sign that it is not yet supported in my device?

melberj, 2025-05-26:
It could be the case that another application is attempting to reserve one or more columns of the NPU. Therefore, it might not be possible to reserve all 5 columns as the `npu1` device specifies.

andrej, 2025-05-27:
Hi @antonio-fc, I am going to try to reproduce your issue. Could you please also share your host-side code (`test.cpp`) that runs your design? That way, I can make sure I run exactly the same code as you.

antonio-fc, 2025-05-27:
Here is the project attached below. I simplified the code a bit, but the error happens all the same. The device type can be changed with the 'device_' variable in 'aie2.py' and the project is run with 'make run'.

[app.zip](https://github.com/user-attachments/files/20463382/app.zip)

andrej, 2025-05-27:
I was not able to reproduce your issue. Running your code (after slight modifications to remove some dependencies, this is what I ran) produces this output for me:

```
./loafty.exe -x build/final.xclbin -i build/insts.bin -k MLIR_AIE --anten 96 --imgsz 128 --iters 10 --warmup 5 --verbosity 0 --verify false
   Avg NPU time: 114597us.
   Min NPU time: 114418us.
   Max NPU time: 115562us.
```

I verified that `device_ = AIEDevice.npu1` in your `aie2.py`, and the line linked by @jgmelber above sets `num_columns = 5` correctly.

This is on a Phoenix-generation NPU, `AMD Ryzen 9 7940HS`, on Ubuntu 22.04.4, MLIR-AIE commit `b101659e08ef092848eb36130219fbebe27c25b3`. What are the details of your setup?

andrej, 2025-05-27:
After upgrading to Ubuntu 24.04, XRT 2.20.0, I can confirm I am also getting the same error as you:

```
terminate called after throwing an instance of 'xrt_core::system_error'
  what():  DRM_IOCTL_AMDXDNA_CREATE_CTX IOCTL failed (err=-28): No space left on device
make: *** [Makefile:91: run] Aborted (core dumped)
```

This makes me think the issue might have to do with an update to either the XDNA driver or XRT. I will investigate further.

andrej, 2025-05-30:
After testing many versions of the `xdna-driver`, I have isolated [`xdna-driver` commit `2019485`](https://github.com/amd/xdna-driver/commit/2019485b933851f3b0da84bc5eb7418d77711f6b) as the commit introducing the issue. As far as I can tell, that commit did not update XRT, so we can rule out XRT as the issue. The commit did add some checks for device metadata, so it could be that these added checks are failing. I will have to see how we can fix this issue.

As a workaround, you can  downgrade your driver to the newest commit before that breaking change. That is [commit `29d7d10`](https://github.com/amd/xdna-driver/commit/29d7d1097f5555c1c1bd1dbc7a70a6d5d4bd6165).

However, since that version of the driver is older, you may need to apply to patches (I needed to do so to get it to work on my machine). In particular, you might need to copy `tools/info.json` out of [`xdna-driver` commit `9be85bb`](https://github.com/amd/xdna-driver/blob/9be85bbbe8e6c343886ee2aa0f5ae959ec19f222/tools/info.json), and in the XRT submodule, cherry-pick [XRT commit `04350f6`](https://github.com/Xilinx/XRT/commit/04350f6857f4bcbaa91c46efd074825c2feb5cf4).

These are the commands with which I was able to get a downgraded driver to work on my machine:
```
# Remove old driver
sudo apt remove xrt-base

# Checkout last working verson of XDNA driver and apply patches so it runs with latest firmware
cd <path to your xdna-driver repository>
git reset --hard
git checkout 29d7d10
cd xrt && git reset --hard; cd ..
git submodule update --recursive
git checkout 9be85bb tools/info.json

# Same for XRT, apply patches so it runs on latest kernel
cd xrt
git cherry-pick -n 04350f6
cd build

# Rebuild XRT
./build.sh -clean
./build.sh -npu -opt

# Install downgraded XRT  (make sure old versions are removed first!)
sudo apt install ./Release/xrt_202510.2.19.0_24.04-amd64-base.deb
sudo apt install ./Release/xrt_202510.2.19.0_24.04-amd64-base-dev.deb

# Rebuild XDNA
cd ../../build
./build.sh -clean
./build.sh -release
./build.sh -package

# Install new XDNA driver
sudo apt install ./Release/xrt_plugin.2.19.0_ubuntu24.04-x86_64-amdxdna.deb
```

After this downgrade, your code once again runs on my end:

```
(wheelenv) andre@andre-vivobook:~/mlir-aie/issue-2333/device_app$ make run
./loafty.exe -x build/final.xclbin -i build/insts.bin -k MLIR_AIE --anten 96 --imgsz 128 --iters 10 --warmup 5 --verbosity 0 --verify false
   Avg NPU time: 115084us.
   Min NPU time: 114573us.
   Max NPU time: 115299us.
```

andrej, 2025-05-30:
I think it might be a bug, or at least a regression of the available functionality.

I don't know what the "partitioning" introduced in that commit does, but my hunch is that in order to partition the device, they need to rely on all columns being equal. The parts of that commit relevant to us add a run queue `start_col`, which on AIE2 is initialized to a value of `1`, and then `total_cols` is computed as the device number of columns minus the start column. Setting `start_col=0` (somehow) would fix the linked check, but I fear this means the partitioning would break.

I got in touch with the author of the commit, and I'll update you once I hear back. Maybe there's an easy way around this (e.g. somehow disable partitioning).

andrej, 2025-06-04:
@antonio-fc After talking to a XDNA driver developer, it appears that moving forward, using the 0-th column will no longer be officially supported by the driver. The only solution if you need that column will be the driver downgrade workaround described above.

closed 2025-06-04 by andrej