Xilinx/mlir-aie #2336: Question: Why use MLIR-AIE and not Vitis AIE applications

flxinxout, 2025-05-23:
Hey,

AMD provides software tools for AIE kernels development. A simple question is why would we use MLIR-AIR toolchain? Maybe I misunderstood one of the purpose of this repo, and something is completely new with it

Thanks
Have a nice day

melberj, 2025-05-26:
Hi @flxinxout

AMD does provide software tools in [Vitis AI](https://www.amd.com/en/products/software/vitis-ai.html) for AIE kernels (xchesscc) and multi-core AIE designs ([aiecompiler](https://docs.amd.com/r/en-US/Vitis-Tutorials-AI-Engine-Development/Vitis-Tutorials-AI-Engine-Development-XD100)), additionally the [Ryzen AI Software](https://www.amd.com/en/developer/resources/ryzen-ai-software.html) package provides tools for deploying AI inference on Ryzen AI PCs. 

The purpose of this repository and [llvm-aie](https://github.com/Xilinx/llvm-aie) is to provide open-source toolkits that target AI Engines in both Versal and Ryzen AI products. This enables enthusiasts and researchers to build their own toolchains for AIEs using MLIR or LLVM. Additionally this repository contains the IRON API and Python libraries for building applications targeting Ryzen AI NPUs. IRON's goal is to provide a close-to-metal interface for fast and efficient programming, this is a lower level of abstraction compared to the ONNX/PyTorch AI inference interfaces in the Ryzen AI Software package mentioned above. 

TLDR: This repository is an extensible open-source toolchain for AI Engines and includes IRON a close-to-metal Python API focused on enabling developers on Ryzen AI NPUs. 

I hope this helps and thanks for your interest!

flxinxout, 2025-05-27:
Thanks @melberj for the explanation.
Does it mean that for Ryzen AI NPUs, with either AIE or AIE-ML architecture, the only frameworks available for programming is MLIR-AIE and Ryzen AI Software? I'm actually summarizing the tools and documentation for programming AI engines.

melberj, 2025-05-27:
Yes. However no Ryzen AI NPUs are AIE architecture, just XDNA(AIE-ML) and XDNA2 architectures.

closed 2025-05-30 by flxinxout